philipptrenz/photo-booth #1: Adding voice control

evancohen, 2017-08-25:
Hey Philipp, just gauging some interest here: would you be interested in merging a PR to add voice control for the photo booth? I'm thinking instead of tapping a screen you say "cheese" and then the countdown starts.

philipptrenz, 2017-08-25:
Hey Evan, many thanks for your support! 
This sounds great, I'm very open minded to add new features! Have you already implemented something?

evancohen, 2017-08-26:
I haven't actually written the code for this yet, but I wrote a node module for an offline keyword spotter that I use in my smart mirror - would be easy to implement something pretty quickly :)

I'll take a stab this weekend.

philipptrenz, 2017-08-27:
Working offline is quite important, I'm looking forward to hear from you ;)

philipptrenz, 2017-11-27:
Hi @evancohen any progress?

evancohen, 2017-11-30:
This totally slipped off my radar! I've got some time tonight and I'll take a stab.

closed 2018-02-26 by philipptrenz

wikijm, 2018-06-13:
@evancohen Any news about this?

evancohen, 2018-06-14:
@wikijm I got a prototype working, but I'm traveling for the next few months without access to my dev box.

wikijm, 2018-06-14:
@evancohen 
OK, thanks for your feedback.

Enjoy your trip 😃