philipptrenz/photo-booth #2: Problems with installation or Raspbian Jessie and Strech

ltsirov, 2017-09-11:
HW used: Raspberry Pi 3 Model B

SW Used: 2017-04-10-raspbian-jessie-lite
SW Used: 2017-08-16-raspbian-stretch

Sequence:
1. Write image to SD card
2. Enable UART and SSH
3. sudo apt-get update
4. Follow the quide on the readme.md of the project

Expected behavior:
5. Installation finishes without problems

Actual behavior:
5. Installation of electron with npm fails on both versions of Raspbian at this step:

# Install dependencies and run the app
npm install && ./node_modules/.bin/electron-rebuild

Attaching the logs for both.
[log_jessie.txt](https://github.com/philipptrenz/photo-booth/files/1292096/log_jessie.txt)
[log_strech.txt](https://github.com/philipptrenz/photo-booth/files/1292097/log_strech.txt)

P.S. Any ideas/suggetions on how to get this working are welcome. I also have RPi1 and RPi2. Timely answers will be greately appreciated, as I have limited time to get this working for my wedding this weekend (or I'm going to get yieled at by my bride :) ). Thanks in advance.

philipptrenz, 2017-09-11:
Hey @ltsirov 

unfortunately at the moment I don't have the time to look over it.  But I can offer you to send my RPi3 Image (starting photo-booth directly upon xserver, with hotspot configuration).

When do you need it working?

ltsirov, 2017-09-12:
Hi @philipptrenz,

Thank you for the fast resposne.

I think getting the image and trying it should be great, or at least it wouldn't do any harm. I was planning to setup the same configuration with hotspot.

I need it working for Saturday, but I would prefer to have couple of days to play and tweak the camera.

BR,

HenHan85, 2017-09-13:
Hello @philipptrenz ,

I am (most likely)  facing the same issue as described by @ltsirov .

Can you tell me which Raspbian image version you used that worked with your instructions?

I am planning to upgrade my DIY photobox with your software for a wedding of a friend in two weeks. 
At the moment we are using an Ipad with a Wifi SDcard in the Camera to download and show the taken Photo, which is unfortunately very slow and unreliable.

I would be very happy to get your feedback in this matter.

Thanks in advance and best regards,

Hendrik

JddAndrewLauren, 2017-09-14:
I am also getting the same issue with both Raspbian Jessie and Stretch.

svargas93, 2017-09-14:
wow i thought i was the only one with this issue.if i could get a copy of the image that would be great and thank for your time.

HenHan85, 2017-09-14:
Hi Philipp, a working image would be great to have.

philipptrenz, 2017-09-17:
Hey guys,

I'm working on it. A solution for this issue and also a sd card image I can share with you. My current image is 16GB, so I have to minimize the size. I'm quite busy at the moment, it will take some time. 

I will share my installation specs (Raspbian, Node and NPM version) with you, I have to look them up.

cgscott903, 2017-09-17:
Hi Philip, add me on for peeps having install probs with Stretch.  :)

olivej, 2017-09-18:
Hi guys, I had the same problem, so I did the following :

1.  Upgrade node using :

curl -sL https://deb.nodesource.com/setup_7.x | sudo -E bash -
sudo apt-get install -y nodejs

2.  Now, in the photo-booth directory, open package.json and where it lists the “electron” packages, remove the specific version number in quotes and put an asterisk between the quotes.  Save the file.

3.  You should be ready to run “npm update” and then do the electron-rebuild

I hope this helps.  This was on the latest Raspbian image.  Don't bother wasting time with the "lite" image, as you'll need quite a few X11 and GTK packages anyway.

cgscott903, 2017-09-19:
Well olivej,  

It looks like it got farther this time but it still failed.  Output is attached.

[error.txt](https://github.com/philipptrenz/photo-booth/files/1314592/error.txt)

JddAndrewLauren, 2017-09-19:
@olivej 's solution helped me out. Upgrading Node allowed Photo Booth to install correctly, but I was getting an error running the app that it had been compiled against a different node.js version. I put the specific version numbers back in the package.json and re-ran the electron-rebuild and I was able to get it working. 

I hadn't started from scratch, so it's quite possible that my node.js issues were related to my own meddling.

olivej, 2017-09-19:
@JddAndrewLauren , very glad it helped move things further - great to know that it can work with the specific numbers in some circumstances.

@cgscott903 , just to give you a bit more insight into the order that I did things, I first installed Raspbian directly from their site (not the lite image).  Then, I removed the original node using apt-get, and then upgraded node using the steps above.  Then, I rebooted to make sure all my environment variables were correct (rebooting on pi is almost as quick as opening new terminal, ha).  Then, I downloaded photo booth, per the instructions on this github.  Then, I changed the version numbers to asterisk for the electron packages.  Then, I kept following the instructions on this GitHub project, and picked up at the step right after downloading.  I didn't make any other changes that I know of, so hopefully this helps.  I looked at your error log above, but I can't figure out what is causing that.  
I did not point out in my original post that I had removed the original node that came with Raspbian, so maybe this will help.  Thanks.

philipptrenz, 2017-09-25:
Thanks @olivej and @JddAndrewLauren for your help!

Setting up the whole node.js pipeline is mostly not as easy as it should be. Changed node.js versions, outdated dependencies etc. are common issues.

I'll look over your fixes and will try to add them to the repo.

philipptrenz, 2017-09-25:
@ltsirov does the above help you fix your issue?

barrettl88, 2017-09-30:
@JddAndrewLauren Did you manage to get your cameras to work with Gphoto2? I can successfully install photo booth I think (sorry, I am a bit of a noob when it comes to coding). but when I start it up it doesn't detect my camera, i'm using a Canon 6D

philipptrenz, 2017-09-30:
@barrettl88 The 6D is listed under http://gphoto.org/proj/libgphoto2/support.php so it should work. 
It seems like a gphoto2 issue.
You should try to update gphoto2 to the newest version and trigger the camera via the terminal commands. Is your camera listed when you execute `gphoto2 --auto-detect`? Could also be important to choose the right setting on your Canon, e.g. try the M-mode.

barrettl88, 2017-09-30:
I've just done a fresh install and it seems to be further than I was before. 

But I get this in the terminal as it starts
 **"[6220:0930/120449:ERROR:CONSOLE(6442)] "Input positions array is not sorted or contains duplicate values.", source: chrome-devtools://devtools/bundled/inspector.js (6442)"**

And it starts the countdown then stops with an oh shit error message. 

I tried the gphoto execution and all i get is this: 
Model                          Port                                            
----------------------------------------------------------

My camera is in manual mode

philipptrenz, 2017-09-30:
@barrettl88 The last shows that gphoto2 does not detect your camera, so it seems to be an issue with gphoto2. You should try to open a issue at https://github.com/gphoto/gphoto2 for that.
I'm pretty sure the first error does not directly correlate with the gphoto2 problem.

barrettl88, 2017-09-30:
Thank you I will try and have a look there. I look forward to using the software :)

barrettl88, 2017-09-30:
So i did a restart and it's now detecting the camera, Hurrah! When I take a photo, it's in black and white on photo booth. The camera isn't set to that.

Would this be due to Gphoto2 or is there something i need to edit?

philipptrenz, 2017-09-30:
@barrettl88 Right, this is a feature of photo-booth, pictures get displayed in grayscale. You can change this by commenting out line 21 in the css/main.css file. From

```css
-webkit-filter: grayscale(100%);
``` 

to 

```css
/* -webkit-filter: grayscale(100%); */
``` 

Then restart photo-booth and pictures should be colored.

cgscott903, 2017-10-02:
Looks like mine is working now.  Thanks all.  Is there an option for preview on the screen before taking a shot?

philipptrenz, 2017-10-03:
@cgscott903 Glad to hear that! gphoto2 does not support a preview like you know it from a camera, like a video. There's a feature in gphoto2 to trigger a preview shot in lower quality, but as far as I know not every camera supports this feature, though from my perspective the benefit of a preview photo is quite low. And when such a feature results in a smaller number of compatible cameras it's not what I want to achieve with photo-booth.
Besides that I think showing the picture without a preview is quite more fun and reminds of the analog times of photography ;)

barrettl88, 2017-10-03:
@philipptrenz Thank you for all your help earlier in the week. Your software is ace! I'm still having trouble with the greyscale. I have managed to get the collage in to colour, but the actual full screen image is still in black and white and I can't see any other line that would help this.

barrettl88, 2017-10-03:
Nevermind, i've just found it, line 118-122 of main.css was causing it to greyscale out as well :)

philipptrenz, 2017-10-03:
@barrettl88 You're right, I forgot about this. sorry!
I'm glad you like it ;)

philipptrenz, 2017-10-03:
@ltsirov  I think the provided tips from @olivej  and @JddAndrewLauren should fix your issue. I close this issue for clarity purposes, if you still don't get it running feel free to open another issue.

closed 2017-10-03 by philipptrenz