philipptrenz/photo-booth #3: main photo too big

barrettl88, 2017-10-05:
Hello @philipptrenz sorry I've had to create an issue to contact you.

When I take a photo with out the console, the main image is cropped down at the bottom as if the image is too wide for the screen.

Is there any part of the code that will resize the main image so it will fit? Even if I add a margin.

Thanks in advance

philipptrenz, 2017-10-05:
Hey,

it's all right. If I get you right you talk about the meant-to-be fullscreen image.
The preview image wrapper is defined in line 125 in css/main.css and get's loaded as background image of the div with id `#preview`. So just change the values of this div.

Did this happen at your default installation or have you tried to customize it?

barrettl88, 2017-10-05:
Thanks for getting back to me, I changed the resolution on my pi which has helped a bit, I hadn't changed anything on it this is the code:

position: fixed;
    padding:0;
    margin:0;

    top:0;
    left:0;

    width: 100%;
    height: 100%;

    z-index: 10000;
    background-size: cover;
    background-repeat: no-repeat;

	animation: fadein 0.5s;
	animation-timing-function: ease-in;
}

philipptrenz, 2017-10-05:
Okay, maybe it's just a pixel error. You can try to set width and height to 99.9%. If this does not help try the same on `#countdown` at line 104 and 105. If both will not fix the issue, maybe a screenshot would help me to understand the problem a bit clearer.

philipptrenz, 2017-11-27:
Hey @barrettl88 could you solve it?

barrettl88, 2017-12-06:
Hello Phillip, I don't think I did, I think I changed the resolution in the end.

philipptrenz, 2017-12-15:
Okay, but you found a workaround for your case, right?

closed 2018-02-26 by philipptrenz