philipptrenz/photo-booth #4: browser is blank and no photos are captured

rock-meister, 2017-10-10:
When I do 
$ sudo npm start
window settings: {"fullscreen":false,"width":1440,"height":900,"backgroundColor":"#000000"}
[1743:1010/085859:ERROR:CONSOLE(6442)] "Input positions array is not sorted or contains duplicate values.", source: chrome-devtools://devtools/bundled/inspector.js (6442)

Observation: The browser is blank and when I click it, starts the countdown then stops with an oh shit error message.

Setup
RPi3 running Raspian 4.9.53
gphoto2 - 2.5.14 (Updated gphoto2 with https://github.com/gonzalo/gphoto2-updater to the latest)

Per the recommendation from @olivej and @JddAndrewLauren on upgrading node, I ran the following command
sudo apt-get install git && git clone https://github.com/audstanley/NodeJs-Raspberry-Pi-Arm7 && cd NodeJs-Raspberry-Pi-Arm7 && chmod +x Install-Node.sh && sudo ./Install-Node.sh;

Now I see that Node is upgraded to the latest stable version
node - 8.6.0
npm- 5.3.0
electron - ???  I don't know how to get the version of electron running.

Camera - Canon 6D
and I am able to successfully run 
$ gphoto2 --summary
Camera summary:                                                                
Manufacturer: Canon Inc.
Model: Canon EOS 6D
  Version: 3-1.1.6
  Serial Number: 616e734bebfc45959f4e6cd947c1a91d
Vendor Extension ID: 0xb (1.0)

Capture Formats: JPEG
[...]

What am I doing wrong?  Any help is greatly appreciated
Thanks
Rock-meister

hapimp, 2017-10-10:
I have the same issue. Have been struggling with this for some time now. Right now I am trying to set up the photo booth with a Raspbian Jessie image. Will let you know if it works.

hapimp, 2017-10-10:
I have it working now with 2017-07-05-raspbian-jessie.zip image from [here](http://downloads.raspberrypi.org/raspbian/images/raspbian-2017-07-05/) and following the @JddAndrewLauren method (although I did not do the "wget"-part).

However, it ONLY works if I boot to command line with the camera connected and turned on, enter the "photo-booth" directory and run `bash start.sh`. Why this is, I have no idea, and I have not thoroughly tested it, but worth a shot for you @rock-meister, if you feel like it.

I'm using a Canon 70D, by the way.

rock-meister, 2017-10-10:
Ok.  I was able to solve the main problem of picture not showing up in photo-booth.  When I ran 
$ gphoto2 --capture-image
New file is in location /capt0000.jpg on the camera 

which is *not* the card but the sdram.

On looking at the reply from Marcus Meissner on gphoto2 mailing list 

I tried 
$ gphoto2 --set-config capturetarget=card

That didn't work.  When I did
gphoto2 --get-config capturetarget
Label: Capture Target                                                          
Type: RADIO
Current: Internal RAM
Choice: 0 Internal RAM
Choice: 1 Memory card

So I did
$ gphoto2 --set-config capturetarget=1

Now when I run
$ gphoto2 --capture-image
New file is in location /store_00020001/DCIM/100CANON/IMG_5420.JPG on the camera

which is the SDCard.  Now I am able to see the pictures in photo-booth.

Net Net there were two issues
(1) Some Html type issue when launching the photo-booth app - Not fixed
(2) Picture not showing up in photo-booth - Solved by saving the jpg into SDCard

HTH
Rock-meister

philipptrenz, 2017-10-12:
hey @rock-meister, thanks for your testing.

(1): I have admit, that I have no idea what this error could mean. The problem is, that photo-booth uses ECMAScript, which gets compiled to valid JavaScript and the given error lines refers to the compiled JavaScript. So I don't know where to look exactly. Do you notice any problems, which could be related to this error?

(2): I added some explaining words about choosing the right capturetarget at the README, so others can profit from your suffer ;)
Maybe I will collect some FAQ's, this would be a good place for this information.

philipptrenz, 2017-10-12:
Hey @hapimp,

this may be related with some incompatibilities the "new" Raspbian-Desktop PIXEL. I never tested photo-booth there, I always start it direct upon the display server with the script. I not even have a desktop installed, for me command line is enough. The script also should start photo-booth directly, bypassing the windows manager of PIXEL. If it works out for you using the script, it may be the best way for now. I'll try to test photo-booth upon an up-to-date Raspbian version soon.

hapimp, 2017-10-17:
Hi @philipptrenz I used the software this friday on a Jessie installation and it worked just great, thanks! I just want to mention that I am not able to get it working on Stretch even when running with `bash start.sh` from the folder. I just get a black screen with or without the dev console. On a Rpi 3 and Canon 70D.

Anyways, I have limited knowledge but will be happy to assist/test if it makes sense, let me know. Thanks for making this.

philipptrenz, 2017-11-27:
@hapimp: Thanks for your offer! I'm quite sure the problem is the new Pixel desktop which seems to be not compatible with Electron, the base framework. For now I think it's okay to just support Jessie, I updated the README with this hint. I hope I'll get the time to completely refactor photo-booth including the Raspbian Stretch support.

closed 2017-11-27 by philipptrenz

hapimp, 2017-11-27:
@philipptrenz I meant to let you know, but got away from it: I tried again just last week, and everything worked as it was supposed to with latest version of Raspbian w. Pixel desktop. I followed the above mentioned method and started the photo booth with `sudo bash start.sh`. It refused to start without sudo.

I think there was some issues with repos when I initially had problems. It seems to have worked itself out know. Again, let me know if there's anything I can do - the photo booth is great.

philipptrenz, 2017-11-27:
Hi @hapimp I'm glad to hear that, thanks!
I'll add this information to the README file so others can benefit from it 😉