philipptrenz/photo-booth #7: Unexpected token import

ThomasVandenabeele, 2017-11-22:
I installed nodejs on my raspberry pi 3 (nodejs version 8.9.1).
When I start the program I only see a black booth.html window.
In the developer console I see an SyntaxError: Unexpected token import at line 18 of booth.html.

I assume this has something to do with a package version mismatch? Do you know what I did wrong?

philipptrenz, 2017-11-27:
Hi @ThomasVandenabeele which version of Raspbian do you use? There were several problems with Raspbian Stretch (with the new Pixel desktop) reported. The older Raspbian Jessie as well as the Ubuntu seem to run fine.

philipptrenz, 2017-11-27:
I just updated the README for Raspbian Stretch support. I did not test it myself but I was told that it works that way. If it fixes your problem please let me know.

mkomarsr, 2018-02-24:
I'm getting the same thing under Jessie.

# cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
# nodejs -v
v5.12.0
# npm -v
3.8.6

mkomarsr, 2018-02-24:
Rebuilding electron seems to have resolved it for me. Strong work!

philipptrenz, 2018-02-26:
Thanks for the feedback @mkomarsr!

closed 2018-02-26 by philipptrenz