philipptrenz/photo-booth #8: -bash: ./node_modules/.bin/electron-rebuild: No such file or director

Esusino, 2017-11-26:
Hi Philipp,

first of all many thanks for ur work, awesome!
I run into a this following issue when running npm install && ./node_modules/.bin/electron-rebuild on cd/photo-booth

-bash: ./node_modules/.bin/electron-rebuild: No such file or directory

After that I cannot run the app as this error appears:

sh: 1: electron: not found
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! photo-booth@0.0.1 start: `electron .`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the photo-booth@0.0.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2017-11-26T09_12_40_235Z-debug.log

Any suggestions on what I shall look at?
Many thanks

Andrea

philipptrenz, 2017-11-27:
Hi Andrea, thanks a lot!

Seems like the dependencies could not be downloaded correctly. Please ensure you have access to the internet and then execute `npm install` from the project folder. If any errors occure please post them here.

philipptrenz, 2017-12-15:
Hey @Esusino any progress, can I close this issue?

closed 2017-12-25 by philipptrenz