philipptrenz/photo-booth #9: Installation fails

danikuci1, 2017-11-27:
Attempting to run `npm install && ./node_modules/.bin/electron-rebuild` produces numerous warnings and errors on a raspberry pi. Please assist if you can.

Attached is the npm-debug file that was produced.
[npm-debug.log](https://github.com/philipptrenz/photo-booth/files/1507517/npm-debug.log)

danikuci1, 2017-11-27:
well, I got slightly further. the versions of NPM and node that ship with raspi images is a bit old, so I updated it from `node -v`  `v0.10.29` and `npm -v`  `1.4.21` to `v5.12.0` and `3.8.6` respectively. I ran the install again and got a new error:

`npm ERR! Failed at the sharp@0.16.2 install script 'node-gyp rebuild'.`
`npm ERR! Make sure you have the latest version of node.js and npm installed.`
`npm ERR! If you do, this is most likely a problem with the sharp package,`
`npm ERR! not with npm itself.`
`npm ERR! Tell the author that this fails on your system:`
`npm ERR!     node-gyp rebuild`

now when I start the program, all I get is a black screen and cannot interact with it.

philipptrenz, 2017-11-27:
Hi @danikuci1 , good work!
Hm, node-gyp is a nodejs tool to compile dependencies, maybe there went something wrong while updating. Please remove the `node_modules` and also the `$HOME/.node-gyp` folder. 
If that doesn't help you could remove the two folders again and try to install node-gyp globally by using `sudo npm install -g node-gyp`.

danikuci1, 2017-11-27:
thanks for the response.

`sudo rm node_modules/ -r`removed the `node-modules` folders, but i wasn't able to find `.node-gyp` folder anywhere. 

I went on to your second suggestion and ran `sudo npm install -g node-gyp`. this succeeded, and now the `.node-gyp` folder is present in my home directory.

And so I ran `npm install && ./node_modules/.bin/electron-rebuild` again. Unfortunately, it failed at the exact same spot.

`npm ERR! sharp@0.16.2 install: node-gyp rebuild`
`npm ERR! Exit status 1`
`npm ERR! `
`npm ERR! Failed at the sharp@0.16.2 install script 'node-gyp rebuild'.`
`npm ERR! Make sure you have the latest version of node.js and npm installed.`
`npm ERR! If you do, this is most likely a problem with the sharp package,`
`npm ERR! not with npm itself.`
`npm ERR! Tell the author that this fails on your system:`
`npm ERR!     node-gyp rebuild`

[npm-debug.log](https://github.com/philipptrenz/photo-booth/files/1507770/npm-debug.log)

:-(

Any further suggestions?

philipptrenz, 2017-11-27:
Okay, let's try to follow the suggestions. You can update nodejs by`git clone https://github.com/audstanley/NodeJs-Raspberry-Pi-Arm7 && cd NodeJs-Raspberry-Pi-Arm7 && chmod +x Install-Node.sh && sudo ./Install-Node.sh;`

danikuci1, 2017-11-28:
so, that didn't seem to help. that command installed nodejs successfully.
`npm -v`
`5.5.1`
`node -v`
`v9.2.0`

however, i've still got an error when running the install command, alebit a new error:

`npm ERR! Failed at the epoll@0.1.22 install script 'node-gyp rebuild'.`
`npm ERR! Make sure you have the latest version of node.js and npm installed.`
`npm ERR! If you do, this is most likely a problem with the epoll package,`
`npm ERR! not with npm itself.`
`npm ERR! Tell the author that this fails on your system:`
`npm ERR!     node-gyp rebuild`

I did see your comment elsewhere about installing it on stretch. I am, however, not running stretch. I am running jessie.

Any other suggestions?

philipptrenz, 2017-11-28:
Okay, hm. As it is a node-gyp related issue I'd like to redirect you to the guys of nodejs developing node-gyp or sharp. They should know what to do.
https://github.com/nodejs/node-gyp
https://github.com/lovell/sharp

lovell, 2017-11-28:
Given both epoll and sharp are failing to install it's most likely that this machine is missing the dependencies of node-gyp, namely Python 2.7, make and g++ 4.8+. Please see https://github.com/nodejs/node-gyp#on-unix

danikuci1, 2017-11-28:
this system is raspberry pi running Raspbian Jessie - it includes python 2.7, make, and g++.

i did run the `npm install -g node-gyp` command and it installed node-gyp -> `/opt/nodejs/bin/node-gyp -> /opt/nodejs/lib/node_modules/node-gyp/bin/node-gyp.js`
`+ node-gyp@3.6.2`

afterwards, I ran philip's installation again: `npm install && ./node_modules/.bin/electron-rebuild`

it begins to install, but gets caught in a loop, saying it doesnt have permissions to access a directory (see screenshot). I let this run for about 30 minutes or so until the raspi froze.

I've pulled the SD card and re-imaged fresh raspian jessie version and ran through the install and setup again but got the same results

:-(

[image]

lovell, 2017-11-28:
If you're doing anything as root then you might need [npm install --unsafe-perm](https://docs.npmjs.com/misc/config#unsafe-perm)

danikuci1, 2017-11-29:
@lovell I was running it with sudo, since it would fail fast if I just ran npm install, telling me to make sure I had write access.

Alas I was able to get it to work. However, I had to abandon Raspian Jessie and downloaded ubuntu mate [https://ubuntu-mate.org/raspberry-pi/](url)

Once I had loaded the new OS, I updated it and installed all prereqs as the readme prescribes with the exception of NPM; doing `sudo apt-get install npm` was still loading an old version. So instead I installed it from `curl -sL https://deb.nodesource.com/setup_7.x | sudo -E bash - \ `

Once that was completed, I was able to clone the repo and compile the application. Although I'm a bit bummed it doesn't seem to work with the raspberry pi distro, I'm happy it is working.

@philipptrenz - I do have two questions for you: (1) it seems to display pics on the screen in black and white, although they are taken in color and stored on both the camera and file system as color photos. How can I adjust this? 

Also (2) how can I change the number of photos taken at one time? Is there a setting I can adjust so it takes 3 or 4 photos each time it is clicked?

Thanks for your help and for this project.

philipptrenz, 2017-11-29:
@lovell Thanks for your advices, I appreciate that! This is Open Source at it's best! ;)

@danikuci1 I'm glad you got it! I can't say why in your case it will not work an Raspbian Jessie, others (including me) already got it working without mayor problems ... 

(1) The photos get displayed in grayscale by css. You can change this by commenting out line 21 in the css/main.css file. From

```css
-webkit-filter: grayscale(100%);
``` 

to 

```css
/* -webkit-filter: grayscale(100%); */
``` 

Then restart photo-booth and pictures should be colored.

(2) There's no such feature yet. It's also a bit tricky. As you may have noticed it takes quite a while to download the photo from the camera. The used framework to communicate with the camera "gphoto2" triggers the photo and downloads the photo afterwards. That means you would have to wait until the first image is loaded, then the next one gets captured and downloaded (this could be done while the the first image gets shown). In my opinion it's not very intuitive. What do you think?

danikuci1, 2017-12-08:
@philipptrenz If I set the camera to take a modest sized photo (not full resolution) I notice it is pretty fast, only a few seconds from taking the photo until it displays on the screen. If the second one could be taken while the first one displays, I think it would work, If this is as easy as changing a couple of lines, I'm willing to give it a shot. Let me know.

For now, I will close this issue since the original problem is resolved.

closed 2017-12-08 by danikuci1