philipptrenz/photo-booth #13: Starting program returns "Error: cannot find module '../core-js/symbol/iterator'

pearson222, 2017-12-14:
I ran into problems with npm and node.js being out of date but have since resolved that.  Upon starting photo-booth I get the following:

pi@raspberrypi:~ $ cd photo-booth
pi@raspberrypi:~/photo-booth $ sudo npm start

> photo-booth@0.0.1 start /home/pi/photo-booth
> electron .

App threw an error during load
Error: Cannot find module '../core-js/symbol/iterator' (While processing preset: "/home/pi/photo-booth/node_modules/babel-preset-react/index.js")
    at Module._resolveFilename (module.js:455:15)
    at Function.Module._resolveFilename (/home/pi/photo-booth/node_modules/electron-prebuilt-compile/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/photo-booth/node_modules/babel-preset-react/node_modules/babel-plugin-transform-react-jsx/node_modules/babel-runtime/helpers/typeof.js:5:17)
    at Module._compile (module.js:556:32)
    at Object.require.extensions.(anonymous function) [as .js] (/home/pi/photo-booth/node_modules/electron-compile/lib/require-hook.js:77:14)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
A JavaScript error occurred in the main process
Uncaught Exception:
Error: Cannot find module '../core-js/symbol/iterator' (While processing preset: "/home/pi/photo-booth/node_modules/babel-preset-react/index.js")
    at Module._resolveFilename (module.js:455:15)
    at Function.Module._resolveFilename (/home/pi/photo-booth/node_modules/electron-prebuilt-compile/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/photo-booth/node_modules/babel-preset-react/node_modules/babel-plugin-transform-react-jsx/node_modules/babel-runtime/helpers/typeof.js:5:17)
    at Module._compile (module.js:556:32)
    at Object.require.extensions.(anonymous function) [as .js] (/home/pi/photo-booth/node_modules/electron-compile/lib/require-hook.js:77:14)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)

philipptrenz, 2017-12-15:
Hey @pearson222 as you have a new issue, did you found a solution for #10 and I can close it?
Anyways, please post the outputs of `sudo node -v` and `sudo npm -v`, as @danikuci1 mentioned (thanks!)

You can also try to run
```
chmod +x start.sh
sudo start.sh
```

pearson222, 2017-12-15:
node v9.2.1
npm v5.5.1

pearson222, 2017-12-15:
chmod +x start.sh
sudo start.sh

returns: sudo: start.sh: command not found

pearson222, 2017-12-15:
Got it working.  Deleted photo-booth and recloned.  Thank you

philipptrenz, 2017-12-15:
perfect, glad to hear!

closed 2017-12-15 by philipptrenz

pearson222, 2017-12-15:
Does the camera sd card need a specific formatting?  Connecting cam returned an error

pearson222, 2017-12-15:
Close this out I will start a new issue if it arises.  Thanks again

philipptrenz, 2017-12-15:
Short answer: The SD card has to be formatted so that the camera can save the images to it, so this shouldn't be an issue.

What can go wrong is that the camera saves the images to the internal RAM so photo-booth can't download them. Run `gphoto2 --get-config capturetarget`, the number referring to the SD card has to be entered as "captureTarget" value in the config.json file (default is 1).

pearson222, 2017-12-15:
Do I run these commands through the console in the photo-booth program? Or through terminal?

Matt Pearson | Designer
Atomic Props & Effects, ltd.
895 Hale Ave. North
St.Paul, MN 55128
office: 612-331-1335
cell: 651-303-8047
www.atomicprops.com <http://www.atomicprops.com/>

danikuci1, 2017-12-15:
@pearson222 the `gphoto --get-config capturetarget` command would be run from terminal. Depending on your exact setup, you may need to `sudo gphoto --get-config capturetarget`
This will return info on the various targets the camera could be saving the photo to. From there, you can adjust the `config.json` file to choose the correct capture target that explicitly names your camera's memory card. Make sense?