philipptrenz/photo-booth #17: Raspberry does not start anymore

hendrikOtto, 2018-02-08:
Hi!

I downloaded and installed everything as explained. I'm using the newes version of rasperian. I made update and upgrade. The first start worked great as I typed ./start.sh. After reboot of the raspberry I got stuck on a login screen. While booting the raspberry shows a "Welcome to Raspberry PI Desktop" sign which is new to me. Then I get an gui Interface to login. The raspberry seems to accept the password "raspberry" but instead of login in it seems that the routine is broken and the gui get back to the login screen. 

If I type a wrong password the raspberry shows me that the password is not valid. If I get to Crlt+Alt+F1 I'm already logged in but I cannot start a gui with startx

I am a bit confused what I should do because there is no error massage or similar. 

Greetings 
Hendrik

hendrikOtto, 2018-02-08:
I could solve the problem with 

`cd /home/machine
mv .Xauthority .Xauthority.old
touch .Xauthority
chown raspberry .Xauthority

 cd /home/user
 ls -a -lh
 chown pi .Xauthority
 ls -a -lh`

philipptrenz, 2018-02-09:
Hi @hendrikOtto seems like starting the xserver directly collides with already on xserver running Desktop with its login procedure.

Anyway, it is not recommended to start photo-booth via the `start.sh` script when using a desktop environment. I use the Raspbian Lite version without a desktop, so I have to start a xserver on my own to get windows displayed. 

The recommended way to start photo-booth on Desktop systems is `sudo npm start` from the project directory. Maybe I should point that out better ;)

closed 2018-02-09 by philipptrenz