philipptrenz/photo-booth #18: Problems during the installation

leo-rocca, 2018-02-13:
Hi,
I really appreciate the work you've done, @philipptrenz 👍 

But it's more than a week now that I'm struggling with the installation of the software. I've been trying all kinds of Raspbian versions and now I'm trying again with Jessie Lite but I'm failing again.
Would you please help me out?

**What should I install/what sequence should I follow?** I found @JddAndrewLauren 's in #2 was more or less working... but then I started facing problems there, too.

I'm a big noob when it comes to coding so I haven't understood much from the other issues.
Thanks in advance from Italy 🇮🇹

danflynndesign, 2018-02-14:
Hi @leo-rocca I am having very similar problems as you. I have tried three versions of Raspbian (Stretch, Jesse, and even a version of Jessie before the pixel OS) and also Ubuntu mate. 

I have had photo booth boot up with a black screen, and I have had it boot up and freeze with errors.

Right now Ubuntu Mate is giving me a "/usr/bin/env: 'node': no such file or directory" error.

Any help would be greatly appreciated, Thanks!

philipptrenz, 2018-02-21:
I understand your problems and installing with Node on the Pi is not that easy. But unfortunately I can not provide support for time reasons. Therefore, II thought about creating a possibility to support me and the project financially. As a reward for the support there will be a fully configured and ready to run sd card image. 

So stay tuned, I'll publish this on the README within the next days

closed 2018-02-21 by philipptrenz