philipptrenz/photo-booth #19: Photo booth won't take pictures after the first shot, and other odd behavior

landaunp, 2018-03-10:
Hi Philip. First off, thank you for creating this project. It is exactly what I was looking for! 

I am running this on a Raspberry Pi 3, following @JddAndrewLauren instructions. I am using a Nikon D40 with Nikon DX 18-55mm lens. 

When I run the software, turn on the camera, and click, it will take 1 pictures ONLY if the lens autofocus is set to manual. Otherwise it errors out ("Oh shit...")

If I try to take a second picture it errors out. If I turn the camera off and on again, and click, then it will take one picture. Strangely, after the count down, the camera will snap a photo, and the program it will briefly display "Oh shit..." before showing the photo. 

Additionally, I tried removing the SD card I had in the camera, and it stopped taking pictures all together regardless of rebooting, or power cycling the camera. 

Lastly, I've commented out the grayscale in main.css  (`/* webkit-filter: grayscale(100%); */`) but the images and collage are still showing in grayscale. If I look at the photos in photo-booth/content/web, they are in color.

landaunp, 2018-03-11:
Issue 1 resolved itself some how. I'm not quite sure what did it. 

Issue 2 is resolved as well, and I apologize for not completely RTFM.

Issue 3 is also resolved thanks to your update. I got everything working so THANK YOU!

closed 2018-03-11 by landaunp

philipptrenz, 2018-03-11:
😊👍🏼

Charlielannie, 2018-06-24:
We have the same problem( Issue 1). Can you help us solve it?

philipptrenz, 2018-06-25:
@Charlielannie I answered you @  #34