philipptrenz/photo-booth #20: Can't connect to the built in web page via Raspberry Pi IP address

landaunp, 2018-03-24:
Hello again. I've got the photo booth software working on my Raspberry Pi 3. I have configured to use it as a wifi access point per the suggested official method listed here: https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md
I do not have it connected to internet via the ethernet point, and I have not set up bridging.

It is broadcasting a SSID, and I can connect to it from my windows computer and android phone. Viewing the Raspberry Pi system logs (tail -f /var/log/syslog) I can see that my devices are getting assigned IP addresses. 

If I ping the raspberry pi (using the IP 192.168.4.1) I get a response. 
If I open my browser to 192.168.4.1, the wlan0 configured IP address, it times out and does not show the photo booth web page. 

Is there anything I need to do to "turn on" the web page? I'm not sure where to go from here to trouble shoot. Thank you again for providing this software and your time!

landaunp, 2018-03-24:
I don't know if I fixed it but I think I figured out part of the problem. Entering "192.168.4.1" into Chromes search bar tries to search google for it the address. You have to type "http://192.168.4.1". 

Additionally, I set up DNS forwarding to "photo.booth" and on all devices (PC, iPhone, Android) I had to go to "http://photo.booth", as just entering "photo.booth" would try to do a google search.

philipptrenz, 2018-03-24:
Hey @landaunp,

you just have to run photo booth with root privileges. 

Please check the output of the Developer Console and post it here.

philipptrenz, 2018-03-24:
Ah, I just saw your second comment. I know this problems, but there’s nothing we can do because every browser has its own peculiarities ...

closed 2018-03-24 by philipptrenz

landaunp, 2018-03-25:
That's kinda what I figured. Thank you for weighing in.