philipptrenz/photo-booth #21: Can this work with the RPi camera module?

umar14, 2018-04-16:
Hello,

philipptrenz, 2018-04-16:
Hey there,

as photo-booth uses gphoto2 to connect to cameras via PTP, there is currently no support for the PiCam. Maybe this will be a feature in the future.
But please feel free to give it a try. If you open a pull request, then I'm happy to include this feature.

umar14, 2018-04-16:
Thank you.. Closing this now

closed 2018-04-16 by umar14

alopezbu, 2019-05-26:
Hi @philipptrenz 
Have you consider in this time introducing the option of the picas. Here in this project, it is possible to use the picam or a reflex with gphoto2, in case you need somewhere to look how it is done.
https://github.com/reuterbal/photobooth

My knowledge is not enough to merge both projects and implement in yours the option of the picam.

philipptrenz, 2019-05-26:
@alopezbu The project you mention is not using gphoto2 for the picam, though they implement an additional camera interface aside gphoto2. As soon as I have some resources to improve the software, this is on top of my feature list 😉