philipptrenz/photo-booth #22: electron failing

voteforpresident, 2018-04-21:
I have tried to update npm and everything but something is not right with electron.

This is the log:

> @philipptrenz/photo-booth@1.0.0 start /home/pi/photo-booth
> electron .

/home/pi/photo-booth/node_modules/electron-prebuilt-compile/node_modules/electron/dist/electron: error while loading shared libraries: libgtk-x11-2.0.so.0: cannot open shared object file: No such file or directory
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! @philipptrenz/photo-booth@1.0.0 start: `electron .`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the @philipptrenz/photo-booth@1.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-04-21T12_40_06_020Z-debug.log

voteforpresident, 2018-04-21:
And this comes up after compiling.
npm WARN lifecycle @philipptrenz/photo-booth@1.0.0~postinstall: cannot run in wd %s %s (wd=%s) @philipptrenz/photo-booth@1.0.0 electron-rebuild /home/pi/photo-booth
added 743 packages in 383.454s

philipptrenz, 2018-04-21:
Hi there,
seems like there are dependencies missing. Please try `sudo apt-get install libgtk2.0-0`. I‘ll update the installation instructions if this fixes it.
You are on Raspbian Stretch?

voteforpresident, 2018-04-21:
Now another dependency is missing.

root@photobooth:/home/pi/photo-booth# sudo npm start

> @philipptrenz/photo-booth@1.0.0 start /home/pi/photo-booth
> electron .

/home/pi/photo-booth/node_modules/electron-prebuilt-compile/node_modules/electron/dist/electron: error while loading shared libraries: libX11-xcb.so.1: cannot open shared object file: No such file or directory
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! @philipptrenz/photo-booth@1.0.0 start: `electron .`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the @philipptrenz/photo-booth@1.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-04-21T13_12_45_383Z-debug.log

voteforpresident, 2018-04-21:
And yes I'm on Raspbian Stretch.

philipptrenz, 2018-04-21:
It won't if dependencies are missing ;)
Do you want to run photo-booth headless or on pixel?

Anyways, please try to install all these:
```
sudo apt --no-install-recommends install xserver-xorg xserver-xorg-video-fbdev xinit x11-xserver-utils pciutils xinput xfonts-100dpi xfonts-75dpi xfonts-scalable libgtk2.0-0 libxtst6
```

voteforpresident, 2018-04-21:
the pi is headless. But for some reason the headless is not working at all.

pi@photobooth:~/photo-booth $ sudo npm install
npm WARN lifecycle @philipptrenz/photo-booth@1.0.0~postinstall: cannot run in wd %s %s (wd=%s) @philipptrenz/photo-booth@1.0.0 electron-rebuild /home/pi/photo-booth
up to date in 44.342s
pi@photobooth:~/photo-booth $ sudo npm start

> @philipptrenz/photo-booth@1.0.0 start /home/pi/photo-booth
> electron .

/home/pi/photo-booth/node_modules/electron-prebuilt-compile/node_modules/electron/index.js:9
  throw new Error('Electron failed to install correctly, please delete node_modules/electron and try installing again')
  ^

Error: Electron failed to install correctly, please delete node_modules/electron and try installing again
    at Object.<anonymous> (/home/pi/photo-booth/node_modules/electron-prebuilt-compile/node_modules/electron/index.js:9:9)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/home/pi/photo-booth/node_modules/electron-prebuilt-compile/lib/cli.js:4:17)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @philipptrenz/photo-booth@1.0.0 start: `electron .`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @philipptrenz/photo-booth@1.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-04-21T23_44_50_483Z-debug.log

Does it work better on pixel desktop?

philipptrenz, 2018-04-22:
Now I understand.
You can’t run electron without a desktop environment. You have to use at least x11 (installed via the above mentioned dependencies) and run photo-booth with 
```
sudo startx /home/pi/photo-booth/node_modules/.bin/electron /home/pi/photo-booth/ -- -nocursor
```

voteforpresident, 2018-04-23:
Unfortunately the headless approach failed. So I tried the raspbian desktop version. The installation went through without any problems. But electron has still some issues.

0 info it worked if it ends with ok
1 verbose cli [ '/opt/nodejs/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v9.11.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle @philipptrenz/photo-booth@1.2.2~prestart: @philipptrenz/photo-booth@1.2.2
6 info lifecycle @philipptrenz/photo-booth@1.2.2~start: @philipptrenz/photo-booth@1.2.2
7 verbose lifecycle @philipptrenz/photo-booth@1.2.2~start: unsafe-perm in lifecycle true
8 verbose lifecycle @philipptrenz/photo-booth@1.2.2~start: PATH: /opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/photo-booth/node_modules/.bin:/usr/l$
9 verbose lifecycle @philipptrenz/photo-booth@1.2.2~start: CWD: /home/pi/photo-booth
10 silly lifecycle @philipptrenz/photo-booth@1.2.2~start: Args: [ '-c', 'electron .' ]
11 silly lifecycle @philipptrenz/photo-booth@1.2.2~start: Returned: code: 1  signal: null
12 info lifecycle @philipptrenz/photo-booth@1.2.2~start: Failed to exec start script
13 verbose stack Error: @philipptrenz/photo-booth@1.2.2 start: `electron .`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:285:16)
13 verbose stack     at EventEmitter.emit (events.js:180:13)
13 verbose stack     at ChildProcess.<anonymous> (/opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:180:13)
13 verbose stack     at maybeClose (internal/child_process.js:936:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
14 verbose pkgid @philipptrenz/photo-booth@1.2.2
15 verbose cwd /home/pi/photo-booth
16 verbose Linux 4.14.34-v7+
17 verbose argv "/opt/nodejs/bin/node" "/usr/bin/npm" "start"
18 verbose node v9.11.1
19 verbose npm  v5.6.0
20 error code ELIFECYCLE
21 error errno 1
22 error @philipptrenz/photo-booth@1.2.2 start: `electron .`
22 error Exit status 1
23 error Failed at the @philipptrenz/photo-booth@1.2.2 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

voteforpresident, 2018-04-24:
With npm start the above and following error occurs.

`0 info it worked if it ends with ok
1 verbose cli [ '/opt/nodejs/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v9.11.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle @philipptrenz/photo-booth@1.2.2~prestart: @philipptrenz/photo-booth@1.2.2
6 info lifecycle @philipptrenz/photo-booth@1.2.2~start: @philipptrenz/photo-booth@1.2.2
7 verbose lifecycle @philipptrenz/photo-booth@1.2.2~start: unsafe-perm in lifecycle true
8 verbose lifecycle @philipptrenz/photo-booth@1.2.2~start: PATH: /opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/photo-booth/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle @philipptrenz/photo-booth@1.2.2~start: CWD: /home/pi/photo-booth
10 silly lifecycle @philipptrenz/photo-booth@1.2.2~start: Args: [ '-c', 'electron .' ]
11 silly lifecycle @philipptrenz/photo-booth@1.2.2~start: Returned: code: 1  signal: null
12 info lifecycle @philipptrenz/photo-booth@1.2.2~start: Failed to exec start script
13 verbose stack Error: @philipptrenz/photo-booth@1.2.2 start: `electron .`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:285:16)
13 verbose stack     at EventEmitter.emit (events.js:180:13)
13 verbose stack     at ChildProcess.<anonymous> (/opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:180:13)
13 verbose stack     at maybeClose (internal/child_process.js:936:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
14 verbose pkgid @philipptrenz/photo-booth@1.2.2
15 verbose cwd /home/pi/photo-booth
16 verbose Linux 4.14.34-v7+
17 verbose argv "/opt/nodejs/bin/node" "/usr/bin/npm" "start"
18 verbose node v9.11.1
19 verbose npm  v5.6.0
20 error code ELIFECYCLE
21 error errno 1
22 error @philipptrenz/photo-booth@1.2.2 start: `electron .`
22 error Exit status 1
23 error Failed at the @philipptrenz/photo-booth@1.2.2 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]`

How should I run the command npm start with `node_modules/.bin/electron main.js` ?

voteforpresident, 2018-04-24:
I was being an idiot, I was still trying to launch it from an ssh session.
With sudo npm start the app starts now without any error. I'm so sorry.

When the app starts the "app window" is just black. I have already increased the gpu memory to 256mb.
And with npm run rebuild I get the error that there is no script named rebuild.

philipptrenz, 2018-04-25:
Never mind 😉
So far so good! Yeah, the last one is my fault. But when the window opens there should not be a electron-rebuild issue.

Please ensure that the developer console is enabled in `config.json` and restart photo-booth and please report if it opens up and if yes which logs are under “console”.
We’ll get it running I promise!

voteforpresident, 2018-04-25:
[https://cl.ly/1x0e08190T2b/Bildschirmfoto%202018-04-25%20um%2018.01.32.png](url)
[https://cl.ly/2r380Y3i3w3B/Bildschirmfoto%202018-04-25%20um%2018.01.44.png](url)
[https://cl.ly/0E0o3J0K1D1U/Bildschirmfoto%202018-04-25%20um%2018.02.22.png](url)

Unfortunately the browser doesn't start/show up either. It's just a black window nothing more.
The CPU Usage is nearly at 0% when the window is open.

voteforpresident, 2018-04-25:
I pulled the git project with the new changes. The program starts now without any errors.
But now the window somehow 'transparent'. When I close it the app gets terminated in the console.

[https://cl.ly/1C3d0d2S2p2D/Bildschirmfoto%202018-04-25%20um%2021.55.01.png](url)

philipptrenz, 2018-04-28:
Did you do a `npm install` after pulling? This behavior is very strange, I'd recommend to delete the `photo-booth` folder and do a fresh clone and install

philipptrenz, 2018-04-28:
Oh and please just use `npm start`, giving the config.json as an argument is not (yet) implemented and I never tested this. It could also be that's the whole problem, so maybe you try this first

voteforpresident, 2018-04-29:
I did a new fresh install. Now I get new errors : (

> @philipptrenz/photo-booth@1.4.3 start /home/pi/photo-booth
> electron .

/home/pi/photo-booth/node_modules/electron/index.js:9
  throw new Error('Electron failed to install correctly, please delete node_modules/electron and try installing again')
  ^

Error: Electron failed to install correctly, please delete node_modules/electron and try installing again
    at Object.<anonymous> (/home/pi/photo-booth/node_modules/electron/index.js:9:9)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/home/pi/photo-booth/node_modules/electron/cli.js:3:16)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @philipptrenz/photo-booth@1.4.3 start: `electron .`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @philipptrenz/photo-booth@1.4.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-04-29T23_01_53_348Z-debug.log

voteforpresident, 2018-04-29:
Found a workaround with sudo npm install --save-dev electron-rebuild --unsafe-perm=true --allow-root

philipptrenz, 2018-05-01:
I could reproduce your problem, it is related to the version of Electron used. Electron has problems running on Raspbian, see electron/electron#10468. Versions with 1.7.x seem not to get patched and 1.8.x does not work either. Due to security vulnerabilities in the 1.7.x branch I had decided to go to the beta version of 2.0. My tests were positive, but apparently this is not consistent for everyone. I'm working on a solution, but as there are still conflicts between different other dependencies and Electron, that's not that easy ...

philipptrenz, 2018-05-01:
I hope I found a solution and would be glad if you can test it.

Therefore please delete the photo-booth folder, clone and install photo-booth again and do a reboot afterwards. Then please try to start photo-booth with the following commands from the photo-booth folder:

```bash
npm start
sudo npm start
node scripts/cli.js
```

I also found out starting photo-booth via SSH will not work, it has to be done from a GUI terminal.

Thanks a lot for your help!

partyblitzer, 2018-05-01:
My booth is working!!! Thank you...

philipptrenz, 2018-05-01:
Very glad to hear! 

@voteforpresident You wrote `script` instead of `scripts` 😉

Both should work fine and in the end they do exactly the same, so it's up to you. 
`npm start` may be a bit handier for manual starts, but you have to be inside the `photo-booth` folder. `node <path to>/photo-booth/scripts/cli.js` can be executed from anywhere, so it's the better option for start scripts.

closed 2018-05-01 by philipptrenz