philipptrenz/photo-booth #23: Uncaught error

tompy94, 2018-04-23:
Install all seemed to go smooth, then on launch I get a black screen with the 'elements console sources' etc tabs and the uncaught error below;

```
Uncaught Error: /lib/arm-linux-gnueabihf/libz.so.1: version `ZLIB_1.2.9' not found (required by /home/pi/photo-booth/node_modules/sharp/build/Release/../../vendor/lib/libpng16.so.16)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/photo-booth/node_modules/sharp/lib/constructor.js:10:15)
    at Object.<anonymous> (/home/pi/photo-booth/node_modules/sharp/lib/constructor.js:256:3)
module.(anonymous function) @ ELECTRON_ASAR.js:173
Module._extensions..node @ module.js:598
module.(anonymous function) @ ELECTRON_ASAR.js:173
Module.load @ module.js:488
tryModuleLoad @ module.js:447
Module._load @ module.js:439
Module.require @ module.js:498
require @ internal/module.js:20
(anonymous) @ /home/pi/photo-booth/node_modules/sharp/lib/constructor.js:10
(anonymous) @ /home/pi/photo-booth/node_modules/sharp/lib/constructor.js:256
Module._compile @ module.js:571
require.extensions.(anonymous function) @ require-hook.js:65
Module.load @ module.js:488
tryModuleLoad @ module.js:447
Module._load @ module.js:439
Module.require @ module.js:498
require @ internal/module.js:20
(anonymous) @ /home/pi/photo-booth/node_modules/sharp/lib/index.js:3
(anonymous) @ /home/pi/photo-booth/node_modules/sharp/lib/index.js:15
Module._compile @ module.js:571
require.extensions.(anonymous function) @ require-hook.js:65
Module.load @ module.js:488
tryModuleLoad @ module.js:447
Module._load @ module.js:439
Module.require @ module.js:498
require @ internal/module.js:20
(anonymous) @ utils.js? [sm]:25
(anonymous) @ /home/pi/photo-booth/app/utils.js:313
Module._compile @ module.js:571
require.extensions.(anonymous function) @ require-hook.js:65
Module.load @ module.js:488
tryModuleLoad @ module.js:447
Module._load @ module.js:439
Module.require @ module.js:498
require @ internal/module.js:20
(anonymous) @ booth.js? [sm]:29
(anonymous) @ booth.js? [sm]:170
Module._compile @ module.js:571
require.extensions.(anonymous function) @ require-hook.js:65
Module.load @ module.js:488
tryModuleLoad @ module.js:447
Module._load @ module.js:439
Module.require @ module.js:498
require @ internal/module.js:20
(anonymous) @ booth.html:inline_0.js:1
```

I have ZLIB 1.2.11 installed (or at least I thought I do), and I don't want to delve to much further in and risk doing some damage. Searching did not give me any answers.

Any ideas?

Thanks in advance

philipptrenz, 2018-04-23:
This may be a build issue or a bug of sharp. I'd suggest to run `npm install` again, maybe something went wrong with rebuilding sharp. If it does not work please post the installation logs.

tompy94, 2018-04-23:
Same uncaught error after running `npm install` again. Not sure where the install logs are?

tompy94, 2018-04-23:
All I see is;

`npm install

> @philipptrenz/photo-booth@1.0.0 postinstall /home/pi/photo-booth
> electron-rebuild

✔ Rebuild Complete
up to date in 62.357s
`

Sorry for being unhelpful, still getting my head around all this.

philipptrenz, 2018-04-24:
I see you are on an older version. There were some problems with dependencies, please update to the latest version (you can use `git pull`). 

Also please tell me your setup, if a Raspberry Pi which Raspbian version and if with desktop or not.

philipptrenz, 2018-04-28:
Edit: Please use `git pull` and reinstall the dependencies afterwards with `npm install`

philipptrenz, 2018-05-03:
any progress @tompy94 ?

wikijm, 2018-05-11:
Hi @philipptrenz,

I have the same issue, with a black screen only.
I have photo-booth 1.4.4 (latest).

BeatDeer, 2018-05-13:
Enable the DevTools and Check your console, i have problems with the `Sharp`, `gPhoto` and `libgphoto2`. I reinstall one by one and fox this problem.

I hope it helps. 😄

philipptrenz, 2018-05-21:
@wikijm Any progress?

tompy94, 2018-05-21:
@philipptrenz 
Sorry for the radio silence, I was building last minute for my wedding and ended up starting from scratch in Python 3 so I knew exactly what was going on. I'm going to post my app soon, not as well written as yours but it got the job done.

I tried a couple different versions, running stretch and jesse, and had no luck, so moved on due to lack of time and wedding commitments. If I get a chance after the honeymoon I'll have another look and see if I can help diagnose.

wikijm, 2018-05-24:
Hi @BeatDeer,

Could you please provide commands to reinstall sharp, gphoto and libgphoto2 ?

philipptrenz, 2018-05-28:
Hey @wikijm, just run `electron-rebuild` to reinstall sharp. For libgphoto2 and gphoto2 use the install script as explained in the install procedure. Anyways, gphoto2 itself is not strictly needed for photo-booth.

philipptrenz, 2018-05-28:
@tompy94 Thanks für the reply, I'm very interested in your approach. I also thought about switching to python3 because of the unstable nature of electron ... But for now I don't have the resources to do so.

wikijm, 2018-05-29:
Hi @philipptrenz,

Thanks for your answer. I won't be able to test before one week (time for me to come back to a wedding), but be sure I will. Don't hesitate to close the issue if you thing it's too long.

philipptrenz, 2018-05-30:
Hi @wikijm,
no problem. Alright, just open a new issue if you still need help.

closed 2018-05-30 by philipptrenz

wikijm, 2018-06-13:
@tompy94 If you're interested, I provide a new solution here: #29 
On my side, I'm interested to see your app 😉