philipptrenz/photo-booth #24: new version does not start

partyblitzer, 2018-04-28:
hi there, the install-process works perfect, but the booth does not start after a reboot. and the folder "/home/pi/photo-booth/content/photos" is missing. My raspberry pi 3 model b is updated und upgraded. do you have a solution?

these are the lines on my screen

~/photo-booth $ sudo npm start

> @philipptrenz/photo-booth@1.4.3 start /home/pi/photo-booth
> electron .

libGL error: MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
using default config.json
window settings: {"fullscreen":true,"width":1440,"height":900,"backgroundColor":"#000000"}
prevent screensaver: true

philipptrenz, 2018-04-28:
I moved the default path for the content folder from i.e. `~/photo-booth/content` to '~/content‘, this seemed to be a better place for me. But you can change this by setting `“content_dir": null,` in `config.json`. Sorry for the trouble.

If you updated from a earlier version please run `npm install` again, I upgraded the electron dependency.

Does the developer console open up if you set `"showDevTools": true,` in `config.json`?

—

partyblitzer, 2018-04-30:
I had completely reinstalled the booth, with the latest scripts. I ran 'npm install' again. but when I run 'sudo npm start' the same error message comes up again. Unfortunately, nothing starts. Even if I enter `showDevTools': true,` in `config.json`.

and 'content/photos' is still missing in 'home' and in 'photo-booth'

philipptrenz, 2018-05-01:
This issue regards to the same problem as #22, please subscribe this one.

closed 2018-05-01 by philipptrenz