philipptrenz/photo-booth #25: Error when saving photo

jasonumd, 2018-05-01:
After getting through the issue 22 that was solved today, I sometimes and able to transfer photos, but mostly I get "Whoops ... Something went wrong when saving the photo, please try again". Using a Nikon D5200.

philipptrenz, 2018-05-02:
When saving the images fails its most likely a problem with the user rights of the `content` folder and the subfolders. I ran into the same problem by sometimes with and sometimes without `sudo`. When photo-booth runs as root all the folders get created with root privileges so that when running afterwards as normal user there are no writing authorization.

May this be the case for your issue?

philipptrenz, 2018-05-02:
They were just for testing purposes under different scenarios.

Easiest way for is to delete the `content` folder and then just following the regular instructions:
https://github.com/philipptrenz/photo-booth#run-photo-booth <https://github.com/philipptrenz/photo-booth#run-photo-booth> <https://github.com/philipptrenz/photo-booth#run-photo-booth>

closed 2018-05-03 by jasonumd