philipptrenz/photo-booth #29: Uncaught Error: /lib/arm-linux-gnueabihf/libz.so.1: version `ZLIB_1.2.9' not found

wikijm, 2018-06-10:
Hi @philipptrenz,

I encounter the quite same issue as #23, which corresponds to the presence of a black screen after manual execution of the command below:
`npm start` or `sudo npm start`

`uname -a` give me this result:

> Linux RaspAP3 4.4.17-v7+ #904 SMP Tue Aug 16 17:02:00 BST 2016 armv7l GNU/Linux

Here are the actions I have carried out and their results:
```bash
pi@RaspAP3:~/photo-booth $ npm install

> @philipptrenz/photo-booth@1.4.4 postinstall /home/pi/photo-booth
> electron-rebuild

✔ Rebuild Complete
up to date in 26.336s
pi@RaspAP3:~/photo-booth $ git pull
Already up-to-date.
pi@RaspAP3:~/photo-booth $ npm install

> @philipptrenz/photo-booth@1.4.4 postinstall /home/pi/photo-booth
> electron-rebuild

✔ Rebuild Complete
up to date in 24.897s
pi@RaspAP3:~/photo-booth $ electron-rebuild
-bash: electron-rebuild : commande introuvable
pi@RaspAP3:~/photo-booth $ npm install --save-dev electron-rebuild
npm notice save electron-rebuild is being moved from dependencies to devDependencies
+ electron-rebuild@1.7.3
added 15 packages and updated 3 packages in 37.45s
pi@RaspAP3:~/photo-booth $ $(npm bin)/electron-rebuild
✔ Rebuild Complete
pi@RaspAP3:~/photo-booth $
```

Then the error I get thanks to the DevTools:
```bash
ELECTRON_ASAR.js:173 Uncaught Error: /lib/arm-linux-gnueabihf/libz.so.1: version `ZLIB_1.2.9' not found (required by /home/pi/photo-booth/node_modules/sharp/build/Release/../../vendor/lib/libpng16.so.16)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/photo-booth/node_modules/sharp/lib/constructor.js:10:15)
    at Object.<anonymous> (/home/pi/photo-booth/node_modules/sharp/lib/constructor.js:259:3)
module.(anonymous function) @ ELECTRON_ASAR.js:173
Module._extensions..node @ module.js:598
module.(anonymous function) @ ELECTRON_ASAR.js:173
Module.load @ module.js:488
tryModuleLoad @ module.js:447
Module._load @ module.js:439
Module.require @ module.js:498
require @ internal/module.js:20
(anonymous) @ /home/pi/photo-booth/node_modules/sharp/lib/constructor.js:10
(anonymous) @ /home/pi/photo-booth/node_modules/sharp/lib/constructor.js:259
Module._compile @ module.js:571
require.extensions.(anonymous function) @ require-hook.js:65
Module.load @ module.js:488
tryModuleLoad @ module.js:447
Module._load @ module.js:439
Module.require @ module.js:498
require @ internal/module.js:20
(anonymous) @ /home/pi/photo-booth/node_modules/sharp/lib/index.js:3
(anonymous) @ /home/pi/photo-booth/node_modules/sharp/lib/index.js:15
Module._compile @ module.js:571
require.extensions.(anonymous function) @ require-hook.js:65
Module.load @ module.js:488
tryModuleLoad @ module.js:447
Module._load @ module.js:439
Module.require @ module.js:498
require @ internal/module.js:20
(anonymous) @ utils.js? [sm]:25
(anonymous) @ /home/pi/photo-booth/app/utils.js:313
Module._compile @ module.js:571
require.extensions.(anonymous function) @ require-hook.js:65
Module.load @ module.js:488
tryModuleLoad @ module.js:447
Module._load @ module.js:439
Module.require @ module.js:498
require @ internal/module.js:20
(anonymous) @ booth.js? [sm]:29
(anonymous) @ booth.js? [sm]:175
Module._compile @ module.js:571
require.extensions.(anonymous function) @ require-hook.js:65
Module.load @ module.js:488
tryModuleLoad @ module.js:447
Module._load @ module.js:439
Module.require @ module.js:498
require @ internal/module.js:20
(anonymous) @ booth.html:inline_0.js:1
```

philipptrenz, 2018-06-12:
Hi @wikijm,

sorry for letting you wait. I have no explanation for this. But it seems to be a issue of sharp rather than photo-booth. Would you mind to create an issue at lovell/sharp? As I was not able to reproduce this issue I can't really give you more than this tip, sorry

wikijm, 2018-06-13:
Hi @philipptrenz,

You gave me an answer in less that 2 days, which seems pretty quick to me for an open source project 😉 
I share the same point of view about the root cause, but it's seems important to get information here for future users 😃 

After few hours of researches, I think that I found the real problem and a solution.

The real problem is your project (at least one of it dependencies) is using ZLIB (Library implementing the deflate compression method found in gzip and PKZIP) with version 1.2.9.

The fact is you may find and install in a easy way versions 1.2.8 and 1.2.11, but not the 1.2.9 (take a look at https://packages.debian.org/stretch/amd64/zlib1g).
To get it, you have to follow steps below:

```bash
# Download zlib-1.2.9 sources
cd ~/; wget https://sourceforge.net/projects/libpng/files/zlib/1.2.9/zlib-1.2.9.tar.gz/download
# Decompress tar.gz file
tar -xvf ~/Downloads/zlib-1.2.9.tar.gz
cd zlib-1.2.9
# Compile sources
sudo -s
./configure; make; make install
cd /lib/x86_64-linux-gnu
# Create symbolic link of version 1.2.9 with libz.so.1 file
ln -s -f /usr/local/lib/libz.so.1.2.9/lib libz.so.1
# Clean temporary install files
cd ~
rm -rf zlib-1.2.9
```

And it's fixed! Be aware that the solution is working for Raspbian and Ubuntu (and I assume with other systems) 😃

If I can find the time to do it, I'll try to implement those steps on your install script 😉

closed 2018-06-13 by wikijm