philipptrenz/photo-booth #32: Canon EOS 1100D

xtremecreative, 2018-06-16:
I have attached now Canon EOS 1100D it takes photo directly from command line but not being detected when I run photo-booth program kindly guide me what to do

philipptrenz, 2018-06-25:
Please check [this](https://github.com/philipptrenz/photo-booth#my-camera-takes-a-photo-but-it-does-not-show-up). If your problem is not caused by this please give me the output of `gphoto2 --show-info` and `gphoto2 --list-cameras`

smaxle, 2018-10-11:
Hi there, i've got the very same problem with the same camera. I changed the capture target but the program will only display the loading animation and then switch back to a black screen with that watermark thing in the left lower corner. The first command doesn't output anything and the output of the second one is in [this text file](https://github.com/philipptrenz/photo-booth/files/2470437/output.txt).

philipptrenz, 2018-10-11:
If the first command does not show anything your camera does not get detected properly. This could be a physical problem, so please check your USB cable.

smaxle, 2018-10-11:
Thanks for your quick reply. I tried multiple cables and `gphoto2 --auto-detect` outputs `Modell                         Port                                            
Canon EOS 1100D                usb:001,030     ` so I think the camera is beeing detectet.

philipptrenz, 2018-10-11:
Then try to get the following command to get a image captured and downloaded. Replace X with the available capture targets for your camera. If this works photo-booth will also work.

`gphoto2 --set-config capturetarget=X --capture-image-and-download`

smaxle, 2018-10-11:
`pi@raspberrypi:~ $ gphoto2 --set-config capturetarget=1 --capture-image-and-download`
`FEHLER: Konnte Bild nicht aufnehmen.`                                      
`FEHLER: Konnte nicht aufnehmen.`

philipptrenz, 2018-10-11:
I'm also German ;) 
So it's no issue of photo-booth, try to change the capture target. If this does not work out create a ticket at https://github.com/gphoto/gphoto2

closed 2018-10-11 by philipptrenz

andi34, 2019-05-31:
I got the same trouble on a 1100D.
It seems to be an old known problem. I somewhere read that there is a problem with the GPhoto Monitor.:
```
sudo rm /usr/share/dbus-1/services/org.gtk.vfs.GPhoto2VolumeMonitor.service 
sudo rm /usr/share/gvfs/mounts/gphoto2.mount 
sudo rm /usr/share/gvfs/remote-volume-monitors/gphoto2.monitor 
sudo rm /usr/lib/gvfs/gvfs-gphoto2-volume-monitor 
```
Rebooted once. Problem solved-