philipptrenz/photo-booth #33: Unable to access webapp

a-disease, 2018-06-22:
Thank you for the incredible project. Works a treat except for some reason I can’t load the photos on my phone. I have set up the hotspot as per the guide mentioned in the readme, but either get error 403 or this page can not be displayed when trying to access it via the ip and port. I’ve tried changing the port to 80, 8081 and the default 8080. But to avail.

philipptrenz, 2018-06-25:
Please post the output of the developer console and I'll see how I can help

m18grunling, 2018-06-26:
Hallo Philipp,

danke für das Projekt! Ich habe auch ein Problem mit dem Webserver :( WLAN HotSpot ist konfiguriert, funktioniert auch, aber mir wird nur die Test-Seite vom Apache angezeigt und nicht die Fotos - was muss ich nur noch tun?

a-disease, 2018-06-26:
Err_connection_refused is the error that appears when the webapp confit is left at the default port 8080

philipptrenz, 2018-06-26:
I think I got the problem. You don’t need to run a Webserver like Apache. photobooth brings its own Webserver and it won’t run when the port is already in use.

m18grunling, 2018-06-26:
oh great, that was the solution:
`sudo /etc/init.d/apache2 stop`

a-disease, 2018-06-27:
Error: listen EADDRINUSE :::80
    at Object.exports._errnoException (util.js:1050:11)
    at exports._exceptionWithHostPort (util.js:1073:20)
    at Server.setupListenHandle [as _listen2] (net.js:1263:14)
    at listenInCluster (net.js:1304:12)
    at Server.listen (net.js:1402:7)
    at Object.<anonymous> (/home/pi/photo-booth/app/webapp_server.js:59:8)
    at Object.<anonymous> (/home/pi/photo-booth/app/webapp_server.js:224:3)
    at Module._compile (module.js:571:32)
    at Object.require.extensions.(anonymous function) [as .js] (/home/pi/photo-booth/node_modules/electron-compile/lib/require-hook.js:77:14)
    at Module.load (module.js:488:32)
(anonymous) @ webapp_server.js? [sm]:39
emitOne @ events.js:96
emit @ events.js:191
window.onerror @ /home/pi/photo-booth/node_modules/electron/dist/resources/electron.asar/renderer/init.js:135

I finally found the developer console which confirms that the address is in use, however even after googling I’m unable to workout which service is using the port. It doesn’t appear to be Apache however as it doesn’t recognise Apache when I run the command mentioned above.

a-disease, 2018-06-28:
Clean install fixed everything, must have been some remnants of another access point I tried to use taking up port 80. Now just to work out dns redirecting

philipptrenz, 2018-06-28:
Congrats @a-disease, you'll also be able to figure out this one ;)

closed 2018-06-28 by philipptrenz