philipptrenz/photo-booth #36: Black screen after pushing the button

m18grunling, 2018-06-30:
...webservice down and photo booth not working :-(

[image]

philipptrenz, 2018-06-30:
Most likely you are running a photo-booth or gphoto2 session anywhere in the background. Do a reboot and make sure you don’t have another instance running.

m18grunling, 2018-06-30:
Hmm.. no.. when i close the application with ALT + F4 and restart.. maybe it is working.

F.e.... i make 5 photos, make a break for 4 minutes and take another photo, then it stop working :-(

ehNorm, 2018-08-16:
I have the same problem with the blank screen, but I do not have this error. Everything seems to be fine and then it suddenly crashes after a while. I guess the app breaks down? Because everything else works fine. Is there any crash log?
CPU or GPU overload?

philipptrenz, 2018-08-26:
@m18grunling The iussue is definitely related to gphoto2, maybe it's a bug related to your camera. Please check (forums etc) if there are known issues with gphoto2 / libgphoto2 and your camera model.

@ehNorm Please open a new issue if it is not directly related to this.

closed 2018-09-26 by philipptrenz