philipptrenz/photo-booth #42: Feature Request- Looping Picture Taking

Tannerj5, 2018-07-24:
In some photo booths, you start the booth by clicking a button, then it automatically takes 3-4 pictures in a sequence without having to push the button again.
I'm a total noob here, but would there be a way to set up the booth.js script so that it would loop the picture taking 3-4 times? This could be configured in the config.json file and turned on or off, and should work with all cameras which are supported.
The basic idea would be to click the button, rund the countdown, take the picture, repeat countdown, take picture 3-4 times. 
Any thoughts?

philipptrenz, 2018-08-26:
It's absolutely possible to do, this should work more or less:

1. Go to `app/booth.js`
2. rename line 80 to `function _trigger() {`
3. write new `function:
```
function trigger() {
    repeat = 5;
    while (repeat-- > 0) {
        _trigger();
    }
}
```

Tannerj5, 2018-08-27:
Hey Phillip- thanks for the reply!

I'm giving your edit a shot today and here's the code I have written up, following your instructions to rename line 80 and writing the function you specified at line 78.
`
function trigger() {
    repeat = 5;
    while (repeat-- > 0) {
        _trigger();
    }
}
const countdownLength = (typeof utils.getConfig().countdownLength == 'number') ? utils.getConfig().countdownLength : 5;

var executing = false;
function _trigger() {`

However, this breaks the trigger entirely and Photobooth doesn't trigger when I click the mouse as usual. I reverted to the original file and Photobooth triggers a single photo as normal. Any ideas on how it could be fixed?

Tannerj5, 2018-09-03:
That works perfectly! Thanks @philipptrenz !!

mikhailbot, 2019-01-03:
Just a small change, the above code will only loop over things once as you're not resetting the counter variable `repeated`. Adjust it the below should remedy that:

```js
if (++repeated < max_repeats) {
  trigger();
} else {
  repeated = 0;
}
```

philipptrenz, 2019-01-07:
@mikhailbot Do you also use this improvement? I'm thinking about integrating it into the project 🤔

mikhailbot, 2019-01-07:
We used it for our wedding this past weekend actually and it was a great success! I think we had nearly 500 photos taken over the course of the night 👍 

I had a USB foot pedal people could use to activate it and we had a cool backdrop wall--the multiple photos being taken was definitely a plus-were able to do multiple poses and silly faces without having to activate it again.

I personally thought people would complain about not seeing a preview before hand but no one seemed bothered too much by it! I had a look at adding it but it wasn't simple so we ran without it.

jgoestl, 2019-03-22:
I implemented the feature request and you can see it in the pull request #66

closed 2019-04-08 by philipptrenz