philipptrenz/photo-booth #43: Push button (GPIO) does not work

c1n4pl3nr3k, 2018-09-30:
Hey there, I have connected a push button (using a 47k resistor) as mentioned in the repository. Also i started NPM with root privileges and turned on the gpio option in my.config.json. Are there any suggestions to fix that?

Kind regards
H
[image]

ComputerGeek586, 2018-10-03:
Also having this issue.

c1n4pl3nr3k, 2018-10-04:
@philipptrenz do you have any suggestion to fix the problem?

Kind regards
H

c1n4pl3nr3k, 2018-10-10:
**@philipptrenz here you find my console.log:**

utils.js? [sm]:235 utils: interface wlan0 192.168.153.148
slideshow.js? [sm]:40 slideshow: enabled
booth.js? [sm]:65 GPIO usage activated
webapp_server.js? [sm]:53 webapp: listening at port 80
camera.js? [sm]:48 gphoto2: Found Canon EOS 400D (PTP mode)

c1n4pl3nr3k, 2018-10-16:
?

philipptrenz, 2018-10-16:
Forgot to hit "comment", sorry!

This seems perfectly fine. Make sure you're using GPIO 3 / PIN 5, I can not give any other advice. I've tested on Raspberry Pi 2 and 3 so it should not be any software related issue.

serpentes80, 2018-10-21:
switched mine a few days ago from Touchscreen to Pushbutton following the guide.
I used a 51kOhm resistor and it works like a charm.
I would check the wiring.