philipptrenz/photo-booth #44: "sudo npm start" Opens to black screen.

ComputerGeek586, 2018-10-03:
When I run the PhotoBooth program with "sudo npm start" a black window opens and nothing happens. When I click the screen with my mouse or press my button connected to pin 3 nothing happens and the camera appears to also be doing nothing. I enabled gpio in the config file so it should be working. Any suggestions? Thanks!

philipptrenz, 2018-10-29:
Any progress here?

philipptrenz, 2018-11-22:
Hi @ZuluForce, thanks for sharing your experience!
It's not the first time a zlib version conflict gets reported (see #23 and #29) and I'm aware of the conflict between sharp and electron. zlib v1.2.8 as well as the v1-branch of Electron which is still currently used for photo-booth has several vulnerabilities. I consider photo-booth to be no security sensitive application, but nevertheless my aim is to upgrade dependencies to unaffected versions. This should also fix the conflicting versions of zlib.
I tried to upgrade dependencies some months ago with the result, that newer versions of Electron won't run on the Raspberry Pi. 
As Electron started semantic versioning beginning with v2.0 and as there are stable releases of it available now I'll risk a new start in the foreseeable future.

philipptrenz, 2018-12-06:
Closing due to inactivity

closed 2018-12-06 by philipptrenz

mlenkeit, 2018-12-11:
For anybody else coming across this issue: I ran into the same problem and specifying `LB_PRELOAD` helped me, too. I didn't try to upgrade the dependencies though.

I ended up changing the `start` script in the `package.json` to include `LB_PRELOAD`:
```json
"start": "LD_PRELOAD=node_modules/sharp/vendor/lib/libz.so electron"
```

philipptrenz, 2018-12-11:
@mlenkeit Thanks!