philipptrenz/photo-booth #45: Image quality

fab31, 2018-10-24:
Hi Philipp,

First of all, I warmly thank you for the job you did on photo booth. 

I would like to know if there is any solution to manage image quality. Most of them after resizing (1500) are less than 100 ko ? I am looking for a better quality equivalent to 1 Mo file. Thank you if you can help me. Fabien

philipptrenz, 2018-10-25:
Hi Fabien,

many thanks!, also for your donation! As the Raspberry Pi has limited resources and disk space I implemented photo-booth so that the high resolution images can stay on the SD card of the camera for later use. 

The images transferred to the Raspberry Pi get shrunk to the `maxImageSize` value (= width in pixels for images in landscape orientation) inside the `config.json`, you can change this value. But be careful, it may influence the performance of photo-booth. 

I hope this helps you out.

philipptrenz, 2018-10-25:
Forgot to mention: The used JPEG quality is 80 and not configurable.

fab31, 2018-10-25:
Thank you Philipp for your quick answer. I have read the code and you confirm me what I suppose. Long life to your project :-) see u

closed 2018-10-25 by philipptrenz