philipptrenz/photo-booth #47: Error on sudo nom start

JayTee2312, 2018-10-29:
Hi,
i got the following message output while starting npm and don't know what to do:

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @philipptrenz/photo-booth@1.4.4 start: `electron .`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @philipptrenz/photo-booth@1.4.4 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-10-29T16_19_08_655Z-debug.log

JayTee2312, 2018-10-29:
0 info it worked if it ends with ok
1 verbose cli [ '/opt/nodejs/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v9.11.2
4 verbose stack Error: Failed to parse json
4 verbose stack Unexpected token < in JSON at position 6 while parsing near '
4 verbose stack
4 verbose stack
4 verbose stack
4 verbose stack
4 verbose stack
4 verbose stack <!DOCTYPE html>
4 verbose stack <htm...'
4 verbose stack     at parseError (/opt/nodejs/lib/node_modules/npm/node_modules/read-package-json/read-json.js:436:11)
4 verbose stack     at parseJson (/opt/nodejs/lib/node_modules/npm/node_modules/read-package-json/read-json.js:104:26)
4 verbose stack     at /opt/nodejs/lib/node_modules/npm/node_modules/read-package-json/read-json.js:51:5
4 verbose stack     at /opt/nodejs/lib/node_modules/npm/node_modules/graceful-fs/graceful-fs.js:78:16
4 verbose stack     at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:525:3)
5 verbose cwd /home/pi
6 verbose Linux 4.14.71-v7+
7 verbose argv "/opt/nodejs/bin/node" "/usr/bin/npm" "start"
8 verbose node v9.11.2
9 verbose npm  v5.6.0
10 error file /home/pi/package.json
11 error code EJSONPARSE
12 error Failed to parse json
12 error Unexpected token < in JSON at position 6 while parsing near '
12 error
12 error
12 error
12 error
12 error
12 error <!DOCTYPE html>
12 error <htm...'
13 error File: /home/pi/package.json
14 error Failed to parse package.json data.
14 error package.json must be actual JSON, not just JavaScript.
14 error
14 error Tell the package author to fix their package.json file. JSON.parse
15 verbose exit [ 1, true ]

philipptrenz, 2018-10-29:
Seems like some files are corrupt. I suggest to delete the folder and clone the repo again

JayTee2312, 2018-10-29:
This is the log file before, after this i downloaded following package.json
https://github.com/philipptrenz/photo-booth/blob/master/package.json

 info it worked if it ends with ok
1 verbose cli [ '/opt/nodejs/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v9.11.2
4 verbose stack Error: ENOENT: no such file or directory, open '/home/pi/package.json'
5 verbose cwd /home/pi
6 verbose Linux 4.14.71-v7+
7 verbose argv "/opt/nodejs/bin/node" "/usr/bin/npm" "start"
8 verbose node v9.11.2
9 verbose npm  v5.6.0
10 error path /home/pi/package.json
11 error code ENOENT
12 error errno -2
13 error syscall open
14 error enoent ENOENT: no such file or directory, open '/home/pi/package.json'
15 error enoent This is related to npm not being able to find a file.
16 verbose exit [ -2, true ]

philipptrenz, 2018-10-29:
I’m not quite sure I understand what you do, but please try to clone the whole repo and start over with the install instructions. If this does not work out please try to uninstall and reinstall node or begin with a fresh Raspbian image

JayTee2312, 2018-10-29:
Thank you for your help, now the start command is functional but now there ist only a black screen

JayTee2312, 2018-10-29:
DevTools was disconnected from page. Once page is reloaded, DevTools will automatically reconnected

JayTee2312, 2018-10-29:
After exiting the browser with alt+f4 following text is visible in the terminal window:

using own config.json
window settings: {"fullscreen":true, "width":1280, "height":1024, "background color":"#000000"}
prevent screensaver: true
### Diese Kamera bietet keine Konfigurationsoptionen.
../../third_party/tcmalloc/chromium/src/tcmalloc.cc:289] Attempt to free invalid pointer 0x6c37f690

philipptrenz, 2018-10-30:
Seems like your camera is not supported, check here: http://gphoto.org/proj/libgphoto2/support.php
Image capture and configuration must be available.

JayTee2312, 2018-10-30:
I found the reason. i thought if i cancel the sdcard connection process in raspbian, the sdcard is not mounted, that's false. If i unmount manually the sdcard your photo-booth software is starting, but it say: no, there is no camera connected. But if you trigger to shot the first photo it will do this, then it is alright.

Now I have to find a way to prevent raspbian to mount the sdcard while starting.

JayTee2312, 2018-10-30:
Now there is another problem.

If I trigger a new photo, the camera takes a photo but photo-booth shows the second time the last shot photo. The new shot photo will be not transferred. If I trigger then the camera the then taken photo will be transferred to photo-booth. I hope you understand what i mean.

philipptrenz, 2018-10-30:
I understand the last comment, the one above not really. 

That the downloaded picture is the one taken beforehand can only be a bug in libgphoto2 in combination with your camera. Please check out the gphoto2 command line tool an check if it behaves the same:

‘gphoto2 —capture-image-and-download’
See: http://www.gphoto.org/doc/manual/ref-gphoto2-cli.html

JayTee2312, 2018-10-30:
Ok, thank you for the really good support. After rebooting the Raspberry all is ok. Shame on me, don't know why i haven't started it before writing. The other comment means that the sdcard was mounted by the os while connecting the camera, i didn't see it before. After unmounting the device /dev/sda1 photo-booth was starting correctly. If the sdcard is mounted, photo-booth is starting with a black screen and nothing happens. I must determine how to prevent Raspbian mounting the sdcard on startup or while plugging in the USB connector. The only thing is the first hint after starting photo-booth **Oh no ... No camera found. Please check the connection and test by triggering a photo.** This message appears every time i start photo-booth but if i trigger a photo it will be taken. it's not really a problem, i know it, so i trigger one photo. But for other users eventually a useful hint. By the way my camera is a Pentax K10D. 
Now i will test the Wifi feature-

Thank you and have a coffee on me ;-)

philipptrenz, 2018-10-30:
Ah, got it: The SD card of your camera gets mounted. I did not experience something like that with other cameras, you can try to create an issue at https://github.com/gphoto/libgphoto2, maybe there's a fix.
Anyways, thanks for the coffee ;)

closed 2018-10-30 by philipptrenz