philipptrenz/photo-booth #48: Hardware Acceleration renders official Raspberry DSI Display black

Knochi, 2018-10-31:
After reboot the display shows the multicolor square and then black screen (backlight is active).

When last line is commented out everthing works as expected.
[config.txt](https://github.com/philipptrenz/photo-booth/files/2533450/config.txt)

philipptrenz, 2018-10-31:
which version of Raspberry Pi do you use?

Knochi, 2018-10-31:
Raspberry Pi 3 B

philipptrenz, 2018-10-31:
Okay. But as the issue is not photo-booth related I ask you to consult relevant forums on Raspberry Pi regarding its hardware acceleration (i.e. https://www.raspberrypi.org/forums/).

closed 2018-10-31 by philipptrenz

Knochi, 2018-10-31:
OK.. a quick google search didn't give any reasonable results. But will Photo-Booth run without GPU acceleration as well? I'm still waiting for the camera cable so I can't try it out right now.