philipptrenz/photo-booth #49: Installation fails on MacOS during electron-rebuild

mlenkeit, 2018-12-05:
Hi community,

on my macOS Mojave MacBook, `npm i` fails during the native electron-rebuild.

I installed vips before via Homebrew but it's still not working. I did a fair amount of searching on Google but couldn't find anything. Xcode is installed, too.

Any pointers are highly appreciated!

Thanks, Max

Error message:

```
> electron-rebuild --verbose

✖ Rebuild Failed

An unhandled error occurred inside electron-rebuild
  TOUCH Release/obj.target/libvips-cpp.stamp
  CXX(target) Release/obj.target/sharp/src/common.o
  CXX(target) Release/obj.target/sharp/src/metadata.o
../src/metadata.cc:86:28: error: use of undeclared identifier 'VIPS_META_IPCT_NAME'
      if (image.get_typeof(VIPS_META_IPCT_NAME) == VIPS_TYPE_BLOB) {
                           ^
../src/metadata.cc:88:43: error: use of undeclared identifier 'VIPS_META_IPCT_NAME'
        void const *iptc = image.get_blob(VIPS_META_IPCT_NAME, &iptcLength);
                                          ^
2 errors generated.
make: *** [Release/obj.target/sharp/src/metadata.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (photo-booth/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:159:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:12)
gyp ERR! System Darwin 18.2.0
gyp ERR! command ".nvm/versions/node/v9.3.0/bin/node" "photo-booth/node_modules/.bin/node-gyp" "rebuild" "--target=1.7.9" "--arch=x64" "--dist-url=https://atom.io/download/electron" "--build-from-source"
gyp ERR! cwd photo-booth/node_modules/sharp
gyp ERR! node -v v9.3.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 

Failed with exit code: 1

Error:   TOUCH Release/obj.target/libvips-cpp.stamp
  CXX(target) Release/obj.target/sharp/src/common.o
  CXX(target) Release/obj.target/sharp/src/metadata.o
../src/metadata.cc:86:28: error: use of undeclared identifier 'VIPS_META_IPCT_NAME'
      if (image.get_typeof(VIPS_META_IPCT_NAME) == VIPS_TYPE_BLOB) {
                           ^
../src/metadata.cc:88:43: error: use of undeclared identifier 'VIPS_META_IPCT_NAME'
        void const *iptc = image.get_blob(VIPS_META_IPCT_NAME, &iptcLength);
                                          ^
2 errors generated.
make: *** [Release/obj.target/sharp/src/metadata.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (photo-booth/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:159:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:12)
gyp ERR! System Darwin 18.2.0
gyp ERR! command ".nvm/versions/node/v9.3.0/bin/node" "photo-booth/node_modules/.bin/node-gyp" "rebuild" "--target=1.7.9" "--arch=x64" "--dist-url=https://atom.io/download/electron" "--build-from-source"
gyp ERR! cwd photo-booth/node_modules/sharp
gyp ERR! node -v v9.3.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 

Failed with exit code: 1
    at SafeSubscriber._error (photo-booth/node_modules/spawn-rx/lib/src/index.js:277:84)
    at SafeSubscriber.__tryOrUnsub (photo-booth/node_modules/rxjs/Subscriber.js:242:16)
    at SafeSubscriber.error (photo-booth/node_modules/rxjs/Subscriber.js:201:26)
    at Subscriber._error (photo-booth/node_modules/rxjs/Subscriber.js:132:26)
    at Subscriber.error (photo-booth/node_modules/rxjs/Subscriber.js:106:18)
    at MapSubscriber.Subscriber._error (photo-booth/node_modules/rxjs/Subscriber.js:132:26)
    at MapSubscriber.Subscriber.error (photo-booth/node_modules/rxjs/Subscriber.js:106:18)
    at SafeSubscriber._next (photo-booth/node_modules/spawn-rx/lib/src/index.js:251:65)
    at SafeSubscriber.__tryOrSetError (photo-booth/node_modules/rxjs/Subscriber.js:251:16)
    at SafeSubscriber.next (photo-booth/node_modules/rxjs/Subscriber.js:191:27)
```

Runtime information:

```
$ vips -v && node -v && npm -v
vips-8.7.1-Mon Nov  5 12:06:40 UTC 2018
v9.3.0
5.5.1
```

philipptrenz, 2018-12-06:
Hey @mlenkeit 

this issue seems to be related to https://github.com/lovell/sharp in combination of electron-rebuild. As first step please try to rerun `npm i`. 

Anyways, the dependencies of electron (and electron-rebuild) are highly outdated because of compatibility issues with the Raspberry Pi on newer versions. That's why I'd like you to update the electron and electron-rebuild dependencies in `package.json`. If you need assistance please feel free to ask, I'll try to help.

mlenkeit, 2018-12-11:
@mlenkeit thanks for the hint. I updated the dependencies to `electron@latest` and `electron-rebuild@latest`. I was then able to successfully start the project on my Mac. However, it didn't recognize my camera then and I switch to a Pi. But I guess the camera issue is unrelated to this. I plan to look into this again this week.

Feel free to close this issue or let me know if you like to to test another config on either/both Mac or Pi.

philipptrenz, 2018-12-11:
@mlenkeit Make sure you have gphoto2 installed on your Mac (or install with `brew install gphoto2`, see [here](http://macappstore.org/gphoto2/)) and your camera is connected properly via USB, then it should work. On an earlier state of development I had photo-booth already running on MacOS, 
only Sierra though.

blak3r, 2018-12-23:
In order to make it install on mac I had to manually remove `rpi-gpio`

### All steps needed
1. `brew install gphoto2`
2. remove this line from package.json ` "rpi-gpio": "^0.9.1",`
3. I deleted package-lock.json (otherwise it still tried to install rpi-gpio)
4. `rm -r node_modules` (if fails try with sudo)
5. `npm install`

### Reason Details

If you do not you'll run into a npm install issue on `etoll` (a dependency) of 

```
Note that epoll can only be installed successfully on Linux systems. Attempting to install epoll on other systems will result in compile errors.
``` 
_Source: <https://www.npmjs.com/package/epoll>_

philipptrenz, 2018-12-25:
Thanks @blak3r for your investigation, I moved the "rpi-gpio" module to `optionalDependencies` so that `npm install` does not fail if the module is not available on other platforms than Linux. 

So for new clonings of the repository or after a `git pull` and `rm -rf node_modules` for the already cloned ones, installation on MacOS is as easy as

1. `brew install gphoto2`
1. `npm install`

blak3r, 2018-12-25:
Great work @philipptrenz

luisdil, 2019-06-24:
Hi community,

I also tried to install photo-booth on macOS (High Sierra) all weekend long but couldn't make it work. At first I also had the problems described by mlenkeit. "npm install" always failed at the rebuild electron process. After I upgraded electron and electron-rebuild it still doesn't work... Then I decided to update all dependencies to their latest version and electron decided to start. Now I am facing the problem that the screen stays black and nothing happens... I don't know what to do next... I tried it with the latest node version and node 10. gphoto2 ist correctly installed via home-brew and works as expected.

Has anyone an idea what I can try?