philipptrenz/photo-booth #50: Is there something to help with this install/run?

csicat, 2018-12-07:


philipptrenz, 2018-12-11:
In which platform? Try running `npm install` again.

Widzwill, 2018-12-15:
Hi Philipp, i appear to run into the same issue -
New Raspberry Pi 3B+ running Rasbian Stretch,  (release 2018-11-13, kernel 4.14)

Nothing installed from the fresh, ran down your installation page one at a time, and it errors on the electron-rebuild during `npm install` 

here's the log file from the failed build -> http://termbin.com/k3p0

philipptrenz, 2018-12-25:
Hi @csicat and @Widzwill, I have another idea where the error could come from.
Can you please change line 12 in `package.json` from `"postinstall": "electron-rebuild"` to `"postinstall":"./node_modules/.bin/electron-rebuild"` and redo `npm install`? 
If this fixes your issue please report it here so I can apply the changes to the repository.

philipptrenz, 2018-12-25:
Hi @Elsurimi 
thanks for your feedback. Please try to also replace `"start": "electron .",` with `"start": "./node_modules/.bin/electron .",`.
I suspect that globally linking the binaries in newer versions of Raspbian will not work for some reason ...

philipptrenz, 2018-12-25:
I updated the repository to provide these relative paths for the "postinstall" and "start" goals, so you can do a `git pull` and try if it fixes this issue.

Elsurimi, 2018-12-26:
Hi @philipptrenz Thanks for your quick feedback.
I downloaded your last version. I still have the same issue:

0 info it worked if it ends with ok
1 verbose cli [ '/opt/nodejs/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v9.11.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle @philipptrenz/photo-booth@1.4.4~prestart: @philipptrenz/photo-booth@1.4.4
6 info lifecycle @philipptrenz/photo-booth@1.4.4~start: @philipptrenz/photo-booth@1.4.4
7 verbose lifecycle @philipptrenz/photo-booth@1.4.4~start: unsafe-perm in lifecycle true
8 verbose lifecycle @philipptrenz/photo-booth@1.4.4~start: PATH: /opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/photo-booth/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle @philipptrenz/photo-booth@1.4.4~start: CWD: /home/pi/photo-booth
10 silly lifecycle @philipptrenz/photo-booth@1.4.4~start: Args: [ '-c', 'electron .' ]
11 info lifecycle @philipptrenz/photo-booth@1.4.4~start: Failed to exec start script
12 verbose stack Error: @philipptrenz/photo-booth@1.4.4 start: `electron .`
12 verbose stack spawn ENOENT
12 verbose stack     at ChildProcess.<anonymous> (/opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:48:18)
12 verbose stack     at ChildProcess.emit (events.js:180:13)
12 verbose stack     at maybeClose (internal/child_process.js:936:16)
12 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
13 verbose pkgid @philipptrenz/photo-booth@1.4.4
14 verbose cwd /home/pi/photo-booth
15 verbose Linux 4.14.79-v7+
16 verbose argv "/opt/nodejs/bin/node" "/usr/bin/npm" "start"
17 verbose node v9.11.2
18 verbose npm  v5.6.0
19 error file sh
20 error code ELIFECYCLE
21 error errno ENOENT
22 error syscall spawn
23 error @philipptrenz/photo-booth@1.4.4 start: `electron .`
23 error spawn ENOENT
24 error Failed at the @philipptrenz/photo-booth@1.4.4 start script.
24 error This is probably not a problem with npm. There is likely additional logging output above.
25 verbose exit [ 1, true ]

philipptrenz, 2018-12-26:
> 12 verbose stack at ChildProcess. (/opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:48:18)

Hi @elsurimi,
did you remove the node_modules folder and did a complete re-install of the project?

juliusheimann, 2018-12-30:
Hi Philipp,

although I just started to install your Software today from the beginning I get the same issues. Did you find the mistake already?

Greetings,

Julius

Elsurimi, 2019-01-02:
Hi @philipptrenz,
Sorry for my delay, I did not have a lot of time to test with Chrismas and the New year.

I have remove node-modules folder and did the complete re-install of the project and it works fine!

thanks for your software and your help

juliusheimann, 2019-01-02:
... same for me. After deleting the folder and a re-install it worked !
Thank you!

closed 2019-01-07 by philipptrenz