philipptrenz/photo-booth #58: App is broken after Out-of-focus exception of gphoto2

philipptrenz, 2019-02-06:
**Description:**  
Everything is working, until some stays to close to the camera and gphoto2 throws an Out-of-focus exception, then your error message appears (Whoops ….), but if you then try to make another photo by clicking left click, the app is broken (dark screen and no reaction). 
On all devices and OS versions i can crash the photo-booth app, if the camera can not do an autofocus.

**Test setup:** 
* Hardware: raspberry pi 3 b+ and tinker board s
* OS: raspbian, armbian stretch, armbian bionic, tinkeros
* Cameras: nikon d3200, d5300

probststefan, 2019-04-06:
I fixed that with `getCurrentWindow().reload();` after the error is thrown.

m18grunling, 2019-04-07:
@probststefan  Where is it necessary to add the line `getCurrentWindow().reload();`

THX!

probststefan, 2019-04-08:
@m18grunling https://github.com/philipptrenz/photo-booth/pull/68

closed 2019-04-08 by philipptrenz