philipptrenz/photo-booth #59: redirect from 80 to 8080

tricktresor, 2019-02-09:
Hi there!
I get an "error listen EADDRINUSE:::80"

you wrote: 

> Basically, it's not a good idea to run a web server as root, if you do not need GPIOs consider setting up a redirect from port 80 to 8080 on your system. That can be achieved by a iptables rule for example.

Can you help me to add the Redirect rule to 8080?

Cheers
Enno

philipptrenz, 2019-02-09:
Hey Enno,

`EADDRINUSE:::80"` means that the port 80 is already in use. Please check if you have any other web server running and if so shut it down, then it should work when you start photo-booth with `sudo`.

If you want to run photo-booth as non-root it will automatically select the port 8080. For redirecting it to the standard port 80 for websites please take a look here:
https://askubuntu.com/questions/444729/redirect-port-80-to-8080-and-make-it-work-on-local-machine

tricktresor, 2019-02-09:
Thanks Philip! I was sure there was no other webserver, but i think there was another photobooth-installation which started apache2 on Startup... :)

cheers
Enno

philipptrenz, 2019-02-09:
So this issue is solved? ;)

closed 2019-02-09 by tricktresor