philipptrenz/photo-booth #60: help with redirect to photo.app

tricktresor, 2019-02-10:
Hi!

I successfully set up the raspberry as access point.
But I am having trouble configuring the redirect to photo.app

I tried to setup DNSMASQ with different options, but none of them worked:
* https://askubuntu.com/questions/53523/how-to-redirect-a-url-to-a-custom-ip-address
* in dnsmasq.conf: 
** address=/photo.app/192.168.4.1"
** local=photo.app/

I have no clue what other things are to do to correctly setup the dns service..,

Do you have a hint for me?

Thanks
Enno

btw: Isn't there a possibilty to automatically call a website after logging on to a network? Like in most hotel access point where you have to accept the rules before using the WLAN? Would this also be possible on raspberry access point?

philipptrenz, 2019-02-12:
Unfortunately I figured out that some mobile devices have problems resolving the DNS entry by dnsmasq. But due to personal time constraints, I can not give you support for DNS redirect. Please have a look at forums regarding this issue.

I already spent a lot time researching regarding the "Captive Portal" feature you mentioned, this would be the ideal implementation from my point of view. 
But I just stumbled over [this project](https://github.com/nodogsplash/nodogsplash) and it looks like it's capable of what you want to achieve, look for the `SplashPage` and `RedirectURL` parameters. If you try it out it would be great if you report your findings!

tricktresor, 2019-02-12:
Cool! Thanks for your Support and hint!
I will try and report!
Enno

closed 2019-02-12 by tricktresor