philipptrenz/photo-booth #62: access the main app via browser

unix0r, 2019-02-23:
Hey guys,
I'm not a web developer and I do not know how Electron works, but I have a customization question.

Is it possible, to access the main app (the main window with slideshow and gallery of all photos) via the built in web browser?

Building my own photo box and I'm struggling which display to use. I think about using an android tablet and put it in some kiosk mode to show the main app.

Thank you for your work. I think its really awesome!

philipptrenz, 2019-02-24:
Hi @unix0r,
the way that Electron works is that the typical tasks of the server (accessing data, using I/O, ...) and the browser (asking the web server for information, rendering the received HTML and running JavaScript) no longer work independently, but instead these are combined in a framework to provide a complete desktop application. Since photo-booth takes advantage of this functionality, it is not that easy to get the Electron UI delivered via a web server to a browser. For this, a lot of functionality would have to be rebuilt and interfaces for communication between server and client had to be implemented.
I hope that answers your question 😉

unix0r, 2019-02-25:
okay thank you.

So i have to look for other possibilities , maybe something like rdp or vnc and streaming the display to the tablet :-)

closed 2019-02-25 by unix0r

vangod90, 2019-02-25:
Look here:
https://www.andrerinas.de/tutorials/raspberry-pi-einen-dslr-weddingphotobooth-erstellen.html

This project has your desired configuration with a Tablet.