philipptrenz/photo-booth #71: Help request

clawitter, 2019-04-14:
Sorry folks, this is in German...

Hallo Philipp,

vielen Dank für das tolle Projekt, das ist genau das, was ich suche!

Ich habe nun mal den Installationsvorgang begonnen, komme aber bei dem Skript nicht fehlerlos zu Ende. Ich benutze ubuntu 18.04 auf einem Laptop.
Hier die Fehlermeldungen aus dem Terminal:

./scripts/install_node_v9.sh: Zeile 7: curl: Befehl nicht gefunden
--2019-04-14 09:24:35--  https://nodejs.org/dist/latest-v9.x/
Auflösen des Hostnamens nodejs.org (nodejs.org) … 104.20.22.46, 104.20.23.46, 2606:4700:10::6814:162e, ...
Verbindungsaufbau zu nodejs.org (nodejs.org)|104.20.22.46|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: nicht spezifiziert [text/html]
Wird in »index.html« gespeichert.

index.html              [ <=>                ]   5,01K --.-KB/s    in 0s

2019-04-14 09:24:36 (25,2 MB/s) - »index.html« gespeichert [5127]

tar: Option benötigt einen Parameter -- „f“.
„tar --help“ oder „tar --usage“ gibt weitere Informationen.
rm: fehlender Operand
„rm --help“ liefert weitere Informationen.
update-alternatives: Fehler: Alternativen-Pfad /opt/nodejs/bin/node existiert nicht
update-alternatives: Fehler: Alternativen-Pfad /opt/nodejs/bin/npm existiert nicht

Danach funktionierte natürlich auch nicht "install npm".

Nun wollte ich fragen, ob Du mir einen Tipp geben könntest, wo der Haken ist? Das wäre echt super!

philipptrenz, 2019-04-14:
Hi @clawitter,

the node install script only is for ARM devices like the Raspberry Pi, please use one of the traditional ways to install node on your laptop. E.g. you can take a look (here)[https://linuxize.com/post/how-to-install-node-js-on-ubuntu-18.04/].

clawitter, 2019-04-14:
Thanks, great!
I will try to install it like that.

clawitter, 2019-04-14:
Strange:
I get "command npm couldn't be found, you can install it with sudo apt install npm".
When I do this, I get "npm is already the newest version (3.5.2)
The same issue with nodejs: "nodejs is already the newest version (8.10.0)".

clawitter, 2019-04-14:
Now I installed it as it is mentioned as an alternative way in your link and it worked.
But then I get another error when running the "npm start" command:

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v9.11.2
4 verbose config Skipping project config: /home/j/.npmrc. (matches userconfig)
5 verbose run-script [ 'prestart', 'start', 'poststart' ]
6 info lifecycle @philipptrenz/photo-booth@1.4.4~prestart: @philipptrenz/photo-booth@1.4.4
7 info lifecycle @philipptrenz/photo-booth@1.4.4~start: @philipptrenz/photo-booth@1.4.4
8 verbose lifecycle @philipptrenz/photo-booth@1.4.4~start: unsafe-perm in lifecycle true
9 verbose lifecycle @philipptrenz/photo-booth@1.4.4~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/jost/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
10 verbose lifecycle @philipptrenz/photo-booth@1.4.4~start: CWD: /home/j
11 silly lifecycle @philipptrenz/photo-booth@1.4.4~start: Args: [ '-c', 'electron .' ]
12 info lifecycle @philipptrenz/photo-booth@1.4.4~start: Failed to exec start script
13 verbose stack Error: @philipptrenz/photo-booth@1.4.4 start: `electron .`
13 verbose stack spawn ENOENT
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:48:18)
13 verbose stack     at ChildProcess.emit (events.js:180:13)
13 verbose stack     at maybeClose (internal/child_process.js:936:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
14 verbose pkgid @philipptrenz/photo-booth@1.4.4
15 verbose cwd /home/j
16 verbose Linux 4.15.0-47-generic
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v9.11.2
19 verbose npm  v5.6.0
20 error file sh
21 error code ELIFECYCLE
22 error errno ENOENT
23 error syscall spawn
24 error @philipptrenz/photo-booth@1.4.4 start: `electron .`
24 error spawn ENOENT
25 error Failed at the @philipptrenz/photo-booth@1.4.4 start script.
25 error This is probably not a problem with npm. There is likely additional logging output above.
26 verbose exit [ 1, true ]

closed 2021-03-30 by philipptrenz