philipptrenz/photo-booth #72: Not launching

ThomasPloechl, 2019-04-15:
Dear Philipp!
After installing and executing the command "sudo npm start" nothing happens.

It only displays 
>@philipptrenz/photo-booth@1.4.4. start /home/pi/photo-booth
> electron .

That's it. No photobooth coming up.
What have I done wrong?

philipptrenz, 2019-05-31:
Hi @ThomasPloechl,

to provide some help, I really need more information than that. Which hardware, which OS (type + version)?

forever80-dev, 2019-07-03:
Hello i have the same problem

Log
---------------------------

0 info it worked if it ends with ok
1 verbose cli [ '/opt/nodejs/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v9.11.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle @philipptrenz/photo-booth@1.4.4~prestart: @philipptrenz/photo-booth@1.4.4
6 info lifecycle @philipptrenz/photo-booth@1.4.4~start: @philipptrenz/photo-booth@1.4.4
7 verbose lifecycle @philipptrenz/photo-booth@1.4.4~start: unsafe-perm in lifecycle true
8 verbose lifecycle @philipptrenz/photo-booth@1.4.4~start: PATH: /opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/photo-booth/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bi$
9 verbose lifecycle @philipptrenz/photo-booth@1.4.4~start: CWD: /home/pi/photo-booth
10 silly lifecycle @philipptrenz/photo-booth@1.4.4~start: Args: [ '-c', 'electron .' ]
11 silly lifecycle @philipptrenz/photo-booth@1.4.4~start: Returned: code: 1  signal: null
12 info lifecycle @philipptrenz/photo-booth@1.4.4~start: Failed to exec start script
13 verbose stack Error: @philipptrenz/photo-booth@1.4.4 start: `electron .`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:285:16)
13 verbose stack     at EventEmitter.emit (events.js:180:13)
13 verbose stack     at ChildProcess.<anonymous> (/opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:180:13)
13 verbose stack     at maybeClose (internal/child_process.js:936:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
14 verbose pkgid @philipptrenz/photo-booth@1.4.4
15 verbose cwd /home/pi/photo-booth
16 verbose Linux 4.19.50-v7+
17 verbose argv "/opt/nodejs/bin/node" "/usr/bin/npm" "start"
18 verbose node v9.11.2
19 verbose npm  v5.6.0
20 error code ELIFECYCLE
21 error errno 1
22 error @philipptrenz/photo-booth@1.4.4 start: `electron .`
22 error Exit status 1
23 error Failed at the @philipptrenz/photo-booth@1.4.4 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

closed 2021-03-30 by philipptrenz