philipptrenz/photo-booth #74: Error Code - An error occurred in the io-library

Hockner, 2019-04-26:
Hi guys,

after new installing photo-booth the app doesn't start any more. The error code is "An error occurred in the io-library" like at the picture attached. 

Could anybody help? Camera is Nikon D3300 or Canon 1300D. Same issue on both cameras.

Thanks!
[image]

lal12, 2019-05-04:
Well, something is blocking the USB Device (as does the "Das Gerät oder die Ressource ist belegt" say). I had the same problem besides killing the mentioned "gvfs-gphoto2" processes, I tried just to look for other processes which could access my usb device.

philipptrenz, 2019-05-31:
Hey, is the problem gone?

andi34, 2019-06-03:
https://github.com/andi34/photobooth/commit/dc1cc9ceb6d77bdb69ef8cbf1e913b8dedd32d93

closed 2021-03-30 by philipptrenz