philipptrenz/photo-booth #77: Print the pictures

baeri3, 2019-05-27:
Hello Philip,

thank you for the nice project, i tested it the last days and it looks realy nice.. 
At the moment i can't find a function to direct print out the pictures on a printer.. Is there a add-on which can do that? Printing with a watermark were great..

Thank you for your attention,
baeri3

philipptrenz, 2019-05-31:
Hi @baeri3,

currently there is no way to connect a printer to photo-booth directly. You could grab the images via the web app and print from your mobile device. Nevertheless the feature of printing the images is often requested and it's on my priority list for the next features.

chico1992, 2019-07-27:
Hi,
I could work on this already have working code,
able to put an d overlay on the downloaded picture and have the ability to print to an ipp compatible printer, tested with a canon selphy 1200 and an HP office jet

philipptrenz, 2019-07-27:
Hey @chico1992, pretty cool! Are you willing to share your code?

chico1992, 2019-07-27:
yes i am just let me know how you want it at the moment everything is hard coded so at the moment still a bit spagetti code 🤣  
should the config be done at the config level one object for the overlay and one for the printer config and if overlay present should it be displayed everywhere or only for the downloaded pictures and printed ones

philipptrenz, 2019-07-27:
Great! 
How does your overlay look like and how does it work? An image with transparency added to the photos?
Two seperate config objects within the `config.json` sound good! First I would add the overlay only to the webapp printed photos, later on there could be an option to choose if it should also be shown within the main display.
Looking forward to test it ;)

chico1992, 2019-07-27:
yes exactly

tigre-bleu, 2019-08-06:
I tested your PR. The print is sent to CUPS but it fails with "The print file could not be opened."

Do you have the same thing?

My printer is a Canon Selphy CP810 with gutenprint driver.

chico1992, 2019-08-06:
hi i still haven't gotten the time to actually test it on my raspberry pi... and didn't have the opportunity to test everything on the pi hopefully i'll get the time this weekend and see if i can help you

tigre-bleu, 2019-08-10:
Seems to be more a problem with gutenprint/cups than with photo-booth itself.

tigre-bleu, 2019-08-11:
There is a problem in the way photo-booth sends the picture to cups. I can print directly using `lp` but not using photo-booth.

For the record, here is a modification of app/utils.js that makes a direct call to the default printer using `lp`. This works for me.

```javascript
  print(filename, grayscale) {
        var _path = path.join(this.photosDir, filename);
        const { exec } = require('child_process');
        exec('lp ' + _path, (err, stdout, stderr) => { 
                if (err) { 
                        console.log("Error tryng to print");// node couldn't execute the command 
                        return; 
                } 
 
                // the *entire* stdout and stderr (buffered) 
                console.log(`stdout: ${stdout}`); 
                console.log(`stderr: ${stderr}`); 
        });
   }
```

marcel-hofer, 2019-09-22:
I'm a little late to this discussion but it looks like you did not finish the printing feature yet.

I also implemented a printing feature along with some other changes (unfortunately this is not a isolated printing-only change).
Compared with the existing one I'm also using CUPS, implemented different layout support (single image, two by one, two by two, ...) and I just added overlay support as in the current implementation.

I tested it and it works with my *Canon Selphy CP1300* in my testing lab. Until now I was not able to test it on a party in *stress mode* but it looks promising.

You find the changes along with a detailed change list in the PR #95
Would be cool if you also consider my approach (along with all other changes) and if not you can probably adopt my *send to printer* implementation.

closed 2021-03-30 by philipptrenz