philipptrenz/photo-booth #79: Liveview

RichyJ999, 2019-06-03:
I see that liveview has now been added to the readme, but isn't working for me on my Canon EOS 1100D. gPhoto says that Liveview is supported on this camera, is there something that I need to enable?

Thanks in advance for any help!

Rich

blaueQuelle, 2019-06-03:
As said in the Readme file, you have to set this setting
`"live": {
		"framerate": 10
	},`
in your my.config.json or config.json. Of course you can set the framerate to a number you prefer

RichyJ999, 2019-06-03:
The "live" option wasn't in the stock config.json and I hadn't noticed it in the readme - thank you for pointing it out!

However with the "live" option now added, I'm still not getting a preview. :( My file now reads:

`{
	"init": {
		"fullscreen": true,
		"width": 1440,
		"height": 900,
		"showDevTools": false,
		"useGPIO": false,
		"grayscaleMode": false,
		"preventScreensaver": true
	},
	"maxImageSize": "1500",
	"firstPhotoCountdownLength": "5",
	"followingPhotosCountdownLength": "4",
	"photoPreviewDuration": "2",
	"photoSeriesLength": "4",
	"slideshow": {
		"enabled": false,
		"activatesAfterSeconds": "30",
		"secondsPerPhoto": "8"
	},
	"gphoto2": {
		"capturetarget": 0,
		"keep": true
	},
	"triggers": {
		"onClick": true,
		"customKeys": ["PageUp","PageDown","b"]
	},
	"content_dir": "../PhotoBooth Photos",
	"webapp": {
		"password": "test",
		"maxDownloadImageSize": 800,
		"enableRemoteRelease": true
	},
	"live": {
		"framerate": 100
	},
	"branding": {
		"type": "text",
		"content": "<div style='font-size: 1.2em; padding-left: 25px;'><i class='fa fa-wifi' aria-hidden='true' style='font-size: 2.5em;'></i> <b style='font-size: 2em; padding-left: 15px;'>photo-booth</b><br /><p>Log into wifi, browse to <b style='padding: 0 5px;'>photo.app</b> and download your photos!</p></div>",
		"position": "bottomleft"
	},
	"flash": {
	  "enabled": false
	}
}
`
I'm running the latest version on a fresh Raspbian install on a RPi3. 

As always, any help is really appreciated! 

Rich

philipptrenz, 2019-06-04:
Hi @RichyJ999, nether your Pi nor your camera will be able to process 100 frames per second, try to stick with the 10 fps, as @blaueQuelle suggested 😉

RichyJ999, 2019-06-04:
Ha ha that's true! I tried values 1, 10, 20, 50 and 100 for the frame rate to see if any would work, but none made a difference.

I assume the rest of the code looks okay?

timetolapse, 2019-06-18:
I have the same issue with EOS 1000D

blaueQuelle, 2019-06-19:
Looks like it as a canon issue? With both my Nikon D700 and D7100 it works out of the box.
Is anything in the console.log?

forever80-dev, 2019-07-03:
I have same problem, the message in console is 
Nikon enable Liveview failed (0xa004: Invalid Status)

The boitier is D3500

lal12, 2019-09-10:
I use it with my Canon EOS 750D. So I don't think it is a Canon issue.

lal12, 2019-09-10:
But it is hard to anything about that, without having proper log output. 

You should set `"showDevTools": true` in `"init"` section, also start it from the terminal. After starting and maybe taking a picture attach the output in the terminal window and in the developer tools console to this issue.

maximilianfriedrich, 2020-02-11:
Make sure your camera is in "manual" mode.

closed 2021-03-30 by philipptrenz