philipptrenz/photo-booth #80: black screen and after reboot no start possible

supi4ever, 2019-06-13:
Hi Philipp,
thx. for the great projekt.
I followed your install instructions and it works fine.
At the firt start of fotobox "npm start" i get a black screen.
Then i used the workariund "export LD_PRELOAD=node_modules/sharp/vendor/lib/libz.so" but it didnt work.
After that i make reboot and tried again, then i got the following error:Error launching app
Unable to find Electron app at /home/pi

Cannot find module '/home/pi'
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @philipptrenz/photo-booth@1.4.4 start: `electron .`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @philipptrenz/photo-booth@1.4.4 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-06-13T17_15_55_556Z-debug.log

Can you help me please to fix this issue?

kind regards
Mario

supi4ever, 2019-06-13:
I reinstalled with fresh raspbian stretch image and updatet.
After installing your code i get black screen already.
I rebooted system and would start npm again anf following error occured :+1: 
pi@raspberrypi:~ $ sudo npm start
npm ERR! path /home/pi/package.json
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall open
npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent 

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-06-13T20_03_57_637Z-debug.log

philipptrenz, 2019-06-13:
You have to execute the commands from within the `~/photo-booth` folder, otherwise it will not work.

supi4ever, 2019-06-14:
Oh ok.
I ececute it from folder and it starts but i cant take a photo with my canon eos70D. After exiting and restarting npm i get following message : 
DevTools was disconnectet from the page.
Once page is reloaded, DevTools will automatically reconnect.
But nothing happens...

philipptrenz, 2019-06-14:
Without more information like hardware, OS, node version, error logs I'm not able to help you.

supi4ever, 2019-06-14:
Yes , sorry !
I use the RPI 3b+ with  Raspbian Stretch with desktop and recommended software Release Date 08.04.2019.
But i think there is a problem with the detection of camera Canon 70D.
Sometimes when i start the photo-booth ther is the black screen with the message " oh no camera found"
In this moment i´m at work , in the evening i can tell you which node version  have installed.

philipptrenz, 2019-06-14:
If it could be an issue with your camera, try the gphoto2 CLI as described here https://github.com/philipptrenz/photo-booth/issues/4#issuecomment-335640172

supi4ever, 2019-06-14:
I tested with another camera eos 400d ang get the following error when i try to capture in terminal:
gphoto2 --set-config capturetarget=1
                                                                               
*** Fehler ***      
An error occurred in the io-library ('Konnte das USB-Gerät nicht beanspruchen'): Konnte Schnittstelle 0 nicht reservieren (Das Gerät oder die Ressource ist belegt). Stellen Sie sicher, dass kein
anderes Programm (gvfs-gphoto2-volume-monitor) oder Kernelmodul (z.B. sdc2xx, stv680, spca50x) das Gerät verwendet und Sie 
Lese- und Schreibrechte für das Gerät haben.

*** Fehler ***      
An error occurred in the io-library ('Konnte das USB-Gerät nicht beanspruchen'): Konnte Schnittstelle 0 nicht reservieren (Das Gerät oder die Ressource ist belegt). Stellen Sie sicher, dass kein
anderes Programm (gvfs-gphoto2-volume-monitor) oder Kernelmodul (z.B. sdc2xx, stv680, spca50x) das Gerät verwendet und Sie 
Lese- und Schreibrechte für das Gerät haben.
*** Fehler (-53: »Konnte das USB-Gerät nicht beanspruchen«) ***      

Für Debug-Meldungen verwenden Sie bitte die Option »--debug«.
Diese Meldungen können Ihnen helfen, eine Lösung für Ihr Problem zu
finden. Wenn Sie jedoch Fehler- oder Debug-Meldungen an die
Mailingliste <gphoto-devel@lists.sourceforge.net> schicken wollen, 
so starten sie gphoto2 bitte wie folgt:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --set-config capturetarget=1

Bitte stellen Sie sicher das die uebergebenen Argument ausreichend gequoted sind.

philipptrenz, 2019-06-14:
See https://github.com/philipptrenz/photo-booth#error--53-could-not-claim-the-usb-device

supi4ever, 2019-06-14:
hmmmm ok , but i dont know which number is the card

supi4ever, 2019-06-14:
i found it ,if you can read you can understand it :-)

For ending the project i have 2 zhings to fix. :- 1 Autostart Photobooth , i didnt find the file /directory "/home/pi/.config/lxsession/LXDE-pi/autostart:" where i have to change the file.
Itś not shown ...

2. When Photobooth starts, on the right side there is a window opened which i have to close every time...

supi4ever, 2019-06-17:
Thx, for you´r support, i think issue can be closed.
It´s solved with "error---53--could.not-claim-tht-usb-device.
Pictures will be taken now.
Thank you ! :-)

closed 2019-06-17 by philipptrenz