philipptrenz/photo-booth #81: Keep the RAWs

FizzyMUC, 2019-06-16:
Hi!
I have my Canon 6D set to take both, RAW plus JPEG. Usually I only do the RAW. Is there any possibility to take both, but only copy the JPEG file to the Raspberry? Seems to always cause an error if there is a RAW file involved.

philipptrenz, 2019-06-16:
Hi @FizzyMUC, this is not possible. 

libgphoto2 does not support to transmit the JPEG image if the camera records RAW+JPEG (see [here](https://github.com/gphoto/libgphoto2/issues/156)). 

I would also really like to add this functionality! Where the feature would have to be added is at the [gphoto2 node wrapper](https://github.com/lwille/node-gphoto2/), feel free to open an issue there.

closed 2019-06-16 by philipptrenz

FizzyMUC, 2019-06-16:
Thanks @philipptrenz for clarification!