philipptrenz/photo-booth #82: Text/butoon overlay

FizzyMUC, 2019-06-16:
Hi all,
I would like to add a button to the main screen (center) that just says sth like “take picture”.
Ideally, the startin point is the image grid, with the button. When I click the button, it goes, away, timer AND liveview start, and then the image is taken.
Sadly I dont know where to start :( any ideas? Is this done in the CSS/HTML?

FizzyMUC, 2019-06-16:
Instead of starting the LivePreview during the initialize function (in booth.js) I have moved it inside the trigger function (before the countdown starts). Now the booth starts with the grid view, however it does not respond to a click/tap on the touchscreen :(

philipptrenz, 2019-06-18:
Hey, this can be achieved by adding just HTML/CSS as the photo sequence can be triggered by clicking anywhere at the screen

FizzyMUC, 2019-06-24:
OK, understood. Will look into it on the weekend.

closed 2021-03-30 by philipptrenz