philipptrenz/photo-booth #83: Change LiveView sequence?

FizzyMUC, 2019-06-17:
Hi all,
I want to have the following User Experience within the application:
Photobooth starts - Opens in grid view - click on display starts the liveview incl. the countdown - image is taken and shown for X seconds - back to grid view (without LiveView active).

I worked on the code a bit and instead of starting the LivePreview during the initialize function (in booth.js) I have moved it inside the trigger function (before the countdown starts). Now the booth starts with the grid view, however it does not respond to a click/tap on the touchscreen :(
Maybe someone has done the same in the past and is aware of a solution?

philipptrenz, 2019-06-18:
Please provide your code changes, e.g. as a PR, so it can be reviewed

FizzyMUC, 2019-06-19:
Will do, as soon as I figure out the "real" GitHub workflow :) Hopefully I can send it tonight.

tigre-bleu, 2019-08-06:
This is an interesting workflow indeed.

closed 2021-03-30 by philipptrenz