philipptrenz/photo-booth #86: Error on sudo npm start

MMatze1984, 2019-07-19:
Hi Philipp,

I've done everything like you described in your manual, but I'm not able to get the app started. It says:

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @philipptrenz/photo-booth@1.4.4 start: electron .
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @philipptrenz/photo-booth@1.4.4 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR! /home/pi/.npm/_logs/2018-07-09T21_27_48_695Z-debug.log

Can you help?

Thanks a lot!

Here is the log:

0 info it worked if it ends with ok
1 verbose cli [ '/opt/nodejs/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v9.11.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle @philipptrenz/photo-booth@1.4.4~prestart: @philipptrenz/photo-booth@1.4.4
6 info lifecycle @philipptrenz/photo-booth@1.4.4~start: @philipptrenz/photo-booth@1.4.4
7 verbose lifecycle @philipptrenz/photo-booth@1.4.4~start: unsafe-perm in lifecycle true
8 verbose lifecycle @philipptrenz/photo-booth@1.4.4~start: PATH: /opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/photo-booth/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle @philipptrenz/photo-booth@1.4.4~start: CWD: /home/pi/photo-booth
10 silly lifecycle @philipptrenz/photo-booth@1.4.4~start: Args: [ '-c', 'electron .' ]
11 silly lifecycle @philipptrenz/photo-booth@1.4.4~start: Returned: code: 1  signal: null
12 info lifecycle @philipptrenz/photo-booth@1.4.4~start: Failed to exec start script
13 verbose stack Error: @philipptrenz/photo-booth@1.4.4 start: `electron .`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:285:16)
13 verbose stack     at EventEmitter.emit (events.js:180:13)
13 verbose stack     at ChildProcess.<anonymous> (/opt/nodejs/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:180:13)
13 verbose stack     at maybeClose (internal/child_process.js:936:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
14 verbose pkgid @philipptrenz/photo-booth@1.4.4
15 verbose cwd /home/pi/photo-booth
16 verbose Linux 4.19.57-v7+
17 verbose argv "/opt/nodejs/bin/node" "/usr/bin/npm" "start"
18 verbose node v9.11.2
19 verbose npm  v5.6.0
20 error code ELIFECYCLE
21 error errno 1
22 error @philipptrenz/photo-booth@1.4.4 start: `electron .`
22 error Exit status 1
23 error Failed at the @philipptrenz/photo-booth@1.4.4 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

MMatze1984, 2019-07-20:
Addition:
I try to run the software on a rPI 3B+ with "Raspbian Buster with desktop and recommended software "(Version: July 2019; Release date: 2019-07-10; Kernel version: 4.19).

I have already set it up again from scratch, but I get the same error...

philipptrenz, 2019-08-08:
Please use Stretch, Buster is not yet tested and Electron is pretty uptight at new Raspbian versions ...

tigre-bleu, 2019-08-09:
I got it working with buster, but had to rebuild some stuff and it was quite painful :)

philipptrenz, 2019-08-09:
> I got it working with buster, but had to rebuild some stuff and it was quite painful :)

Awesome! Can you give some instructions? I'm still waiting for my Pi 4 to arrive 😅

tigre-bleu, 2019-08-09:
It was on a Pi3 because now Buster is the latest raspbian stable. I will check precisely what I did when I'll be home but it involved manually installing electron and sharp from npm then rebuilding sharp due to it being compiled against an incompatible version of node.

tigre-bleu, 2019-08-10:
I had a look in my bash history but it is cluttered a lot as I tried a lot of stuff. The most interesting parts are in addition to what I said above is:
- sudo npm rebuild sharp --update-binary
- sudo ./node_modules/.bin/electron-rebuild

Sorry I don't have a clear tutorial but you might get it working eventually :)

Sralelb, 2019-09-02:
Hi you all,

I have a similar problem with Raspberry 3 and Raspian Strech Desktop. I run installation in the order Philipp points out, but on start command I get the following error message:
[image]

Can anyone help?

Thanks in advance!

tigre-bleu, 2019-09-09:
Well are you sure `package.json` file exists in the `/home/pi/` folder. Can you provide the output of `ls -la /home/pi`?

Sralelb, 2019-09-09:
Well in fact that was the problem. I was in the main folder and not in the photo-booth folder.
So I ran `cd photobooth` first, then `npm start` and it works.

PrinzessinLea, 2020-11-29:
> 
> 
> Please use Stretch, Buster is not yet tested and Electron is pretty uptight at new Raspbian versions ...

Hi! I have exactely the same problem a MMatze. Is it still that I need Stretch instead of Buster? Or has anyone found a solution like tigre-bleu suggested? I've tried that but it didn't work for me.

closed 2021-03-30 by philipptrenz