philipptrenz/photo-booth #97: Live preview PTP Device busy

lefrosch81, 2019-10-03:
Hi,
I tried the new version of lal12 of live preview and I got the following issue. I use a Nikon D5000. Without live preview everything works fine. But with live preview:
The preview starts(Camera does a „Klack“ and the countdown starts, but when the picture will be taken I receive a error message. In terminal it says „PTP Device busy“. The program reloads the startpage an live preview opens again.
So no picture is taken.

DavicVictor, 2019-10-20:
Hi, 

I am having the same problem. 
I am not able to try it now but I think we can solve the problem by adding a timeout between
`if(livePreview)
        livePreview.stop();`
And
`camera.takePicture` 
in booth.js at line 150.

I will get back to you as soon as I am able to try it.

thinkingbox-tien, 2019-11-07:
I'm have he same problem with the Nikon D90 while using liveview. I've done timeouts and tried a while loop to check the camera object to see when it's ready, but when camera.takePhoto triggers, for some reason, that camera object is still undefined.

Any further help would be great!

Tiendongle, 2019-11-07:
I'm have he same problem with the Nikon D90 while using liveview. I've done timeouts and tried a while loop to check the camera object to see when it's ready, but when camera.takePhoto triggers, for some reason, that camera object is still undefined.

Any further help would be great!

closed 2021-03-30 by philipptrenz