philipptrenz/photo-booth #99: Very slow Preview with Raspberry 3B+

ratsplayer, 2019-10-17:
Hi,

i am using Raspberry 3B+ with Raspbian Buster (light and added Pixel) and Canon EOS 1100D.
The preview image is created and shown very slowly and the waiting image is shown a bit slowly.
What do you think could be a good solution to get it work faster?
Do you think i should reinstall it?
Do you think using the new Raspberry 4B will be a good solution?

I also tried live preview and the camera image is in very bad quality.
If i put the camera objective to "auto" the creation of foto takes to long and the software crashes (i am using the pull request with possible gif creation)

Thanks for your help!

timetolapse, 2019-10-17:
Hey,

I am using a 1000D and tried with a 4B with 4 GB of RAM and it didn't help in performance of the live view.

ratsplayer, 2019-10-17:
Hi,

thank you for your answer.
Do you also have performance problems without live view on showing the preview image and showing the waiting image? It is a bit laggy.
I can understand that the live view performance is not perfect on a Raspberry 3B+ but that is ok. 
But the preview image and the loading image could be much quicker i think.

And the image quality of the live view is bad in my case. Of course the flash is off if the live view is shown. But the image of course is slow and the quality is bad so i am thinking about deactivating the live view.

marcel-hofer, 2019-10-19:
Hi

The speed of this action depends on many different factors like:
- Taking picture by camera and saving to sd card (I/O)
- Transfering picture from camera to PI (I/O)
- Saving full resolution image to sd card (I/O) - only when printing feature from PR #95 is enabled
- Resizing image (compute) and saving to disk (I/O)
- Displaying image on the screen (compute + I/O)

I did a few tests using different settings (camera and application). I measured them by hand so the results are not 100% accurate. Also note that additional background tasks (like using the webapp / creating gif / creating print previews or printing) will also slow down the entire processing.

| Image Size | MP | Image Resolution | Image Size | Printing Enabled | Duration | Loading spinner shown |
| --- | --: | --: |  --: |  --- |  --: |  --- |
| Large | 12.2MP | 4288x2848 | 4'300KB | Yes | 3.92s | Yes |
| Large | 12.2MP | 4288x2848 | 4'300KB | No | 3.67s | Yes |
| Medium | 6.9MP | 3216x2136 | 2'375KB | Yes | 3.27s | Yes |
| Medium | 6.9MP | 3216x2136 | 2'375KB | No | 2.68s | No |
| Small | 3.1MP | 2144x1424 | 1'158KB | Yes | 2.68s | No |
| Small | 3.1MP | 2144x1424 | 1'158KB | No | 2.48s | No |

Taking the first picture right after starting *photo-booth* took way longer (possible due to other background tasks still running). In fact it took about 10.51s instead of 3.92s when taking the first picture right after start. The following pictures were much faster (see table above).

I was using a *Nikon D90* with a max resolution of `12.2 MP` which should be pretty similar to your *Canon EOS 1100D*.

Based on the results above and how the application works I suggest considering the following to improve the performance:
- The Raspberry PI is not a high performance PC and thus not optimal for transferring and processing at a high speed (including running an electron app and feeding a 1920x1080p screen). Like @timetolapse also stated it also won't help to upgrade to the newest version.
  I bought an *Intel NUC* a few years ago which is more expensive, a little larger but also faster in every aspect (better CPU and faster single core performance / more RAM / fast SSD / USB 3.0+ for newer cameras). Although I did not test *photo-booth* on it I bet it will be much faster.
- If you don't need the printing feature then disable it. This way the high resolution image won't be saved to the PI for later processing and it will save you a few milliseconds processing time.
- Consider lowering your camera resolution so that less data has to be transferred / processed. My lowest camera setting of 3.1MP is still enough for most applications (also for printing with my low quality printer *Canon Selphy CP1300*). But if you need the full resolution images for later processing this probably won't be an option.

Sure there is always space for code improvements like using faster libraries for image processing and only load full image size when printing but you will get a better performance boost if you consider the points stated above.

To answer all your questions:
- No, reinstalling won't help
- No, Raspberry 4B won't help much - if considering other hardware, then invest in something way better than a PI to really get an inprovement
- I cannot help you with the bad quality of the preview image...If using the live view on the camera the preview is being displayed on a really small screen with typically low resolution. I don't know how *gphoto2* exactly works but if the live view only mirrors this picture (for better performance) then you won't ever have a good image quality for a full hd screen. In this case this will be a hardware limitation if my assumptions are correct.
- I also have problems with using the auto focus of the camera. For me it crashes everytime when the camera is not able to focus correctly and then won't take a picture. In this case the application fails. Well, I don't like the auto focus because it will take about 1 second to find focus before taking the picture and also because it can fail (for example when in low light using flash). I rather try to manually focus (probaly also changing apperture to get more into focus) because it is more reliable. I also would like to see a good solution for that but really don't know how (eg. retry on failure is probably not the way).
- I also had performance issues with the live view. There is an open PR #94 which should improve this and hopefully will work better on the PI. This PR is also on my todo list for testing :-)

ratsplayer, 2019-10-21:
Hi,

thank you for your great answer.

I can accept the slower performance (I don't like it and i will try to improve it by better configurations). But what I really don't like is the laggy loading spinner.
Everybody can understand that creating a photo can take some seconds, but you can see on the screen that it does't run perfectly. It has some lags.
Also the "3" on the counter is not shown in the same duration as the "2" and the "1".

It is possible to run the spinner in another thread or something like that so that it won't run laggy?

philipptrenz, 2019-11-02:
Hey there, I haven’t yet tested PR #94, but in the standard configuration the spinner should not be so laggy. Please check if you have the hardware acceleration of the Pi installed successfully. A reinstall might fix it:

```
# If you are using a Raspberry Pi > 1: Activate hardware acceleration
sudo apt install libgl1-mesa-dri
sudo su -c 'grep -q -F "dtoverlay=vc4-kms-v3d" /boot/config.txt || echo "dtoverlay=vc4-kms-v3d" >> /boot/config.txt'
```

Duesentrieb83, 2019-11-27:
Hi there,
sorry if I post here wrong, but I have also a very laggy perfomance at the countdown.
I'm new to Git, so please be patient if i struggle to follow your instructions to solve the problem.
The "5", "4", "3", "2", "1" are not shown in the same duration.
So far I haven't customized anything at the config.
I also tryed to activate the hardware acceleration as shown in the manual, even if I'm on a rpi3b. But after that npm start ends with a blackscreen. So I fall back to backup Image.
The RPI is on the latest Stretch image that i found in the raspbian archiv.
I also updated the npm to version 6.13.1.
Cam is a Nikon D40 with latest firmware, running in ptp-mode.

maximilianfriedrich, 2020-02-12:
Hi,

there was obviously an issue with hardware acceleration in raspbian-2019-09-30:
[https://github.com/raspberrypi/linux/issues/3391](url)

I had the same thing with this release, after reboot there was only the curser blinking , if `dtoverlay=vc4-kms-v3d` was not commented in the `/boot/config.txt`.

It's definitely working with raspbian-2019-07-12, but also fixed in the newest release raspbian-2020-02-07.

Hope this works for you!

closed 2021-03-30 by philipptrenz