philipptrenz/photo-booth #100: FujiFilm X-T3 Not detected.

Rcordosi, 2019-10-20:
Cannot get X-T3 to work. Get the error that there is no camera detected. Camera is set for PC Connection Mode AUTO. Possibly, the USB port is being taken by another process first?

[my-logfile.txt](https://github.com/philipptrenz/photo-booth/files/3748422/my-logfile.txt)

[summary.txt](https://github.com/philipptrenz/photo-booth/files/3748423/summary.txt)

philipptrenz, 2019-11-02:
Hey, this probably is no photo-booth but a gphoto2 issue. Try to get your camerawomen work using the gphoto2 command line interface. If that works, photo-booth does too.

Rcordosi, 2019-11-04:
I understand. In reality, all Fuji cameras should be removed from the list of acceptable cameras. None of them work. Fuji is very closed about their remote camera operation. I have been trying to get My Fuji X-T1 and X-T3 to work with your program for 3 years. Look around the web, there is no solution. There are many, very long discussions about it and some people have spent many hours trying to program a solution with no luck. Again. Fuji should be taken off the list.

Rob

philipptrenz, 2019-11-05:
Oh okay, wow. Thanks for your investigations and I’m sorry to hear that. 
So I‘ll close this issue, but please feel free to report your findings to the gphoto2 project!

closed 2019-11-05 by philipptrenz