philipptrenz/photo-booth #103: Photos on main screen

rawbertp, 2019-11-02:
I'm not really getting the pattern for selecting the pictures displayed on the "main screen". What I mean, when launching the app there are 3 x 2.5 (on my screen) pictures/tiles displayed. Even though there are far more and more recent pictures it always shows the same 9 pictures.

Expected:
* The most recent photos are shown.
* Or alternatively, a random (changing) selection of photos is shown.

philipptrenz, 2019-11-03:
Hey, the number of shown images depends on your screen size and indeed only the most recent ones get shown. If you take a new one, it gets prepended in the top left corner. A random selection of photos (in fullscreen) can be achieved by activating the slideshow.

rawbertp, 2019-11-03:
Unfortunately, it is not like that. So the pictures displayed are neither the most recent nor the oldest ones. They are somewhere from the last third of pictures and also do not change. Very strange...

vectorez, 2020-02-29:
so that more images are shown, you can change the line 163 in the util.js file

changue this:  var div = $('<div class="img-wrapper col-6 col-md-4">').append(img);

for this: var div = $('<div class="img-wrapper col-2">').append(img);

closed 2021-03-30 by philipptrenz