philipptrenz/photo-booth #115: Photoboth stays black when camera is connected

chrissisdl, 2020-02-13:
Hello, thank you for this program.
A few months ago I installed this photobooth and tried a bit on the side and now wanted to install the program you provided. Now the photobooth only works if no camera is connected, a camera (Canon EOS 500D) is connected because the screen is black and the photobooth does not start.
Has anyone had this problem before?

Sorry for the bad English, I'm from Germany.

I am using a rapsberry 3b +

Warm greetings from Germany

DickesHortKind, 2020-02-15:
Hi chrissisdl,

based on your description, it is of course difficult to find an exact solution.  But maybe it occurs through the automount of the raspi, which prevents the use by CUPS.  I also had this problem.

Maybe this will help you: 
[Link Disable Automount ](https://raspberrypi.stackexchange.com/questions/65739/stock-pi3-what-process-is-auto-mounting-removable-media)

chrissisdl, 2020-02-16:
Hello DickesHortKind, thank you for your answer. I was now able to solve the problem with the black screen after starting the Photobooth. The problem was that there was still a memory card in the camera slot.
In the previous version of the Photobooth, you could leave the memory card in the slot and start the Photobooth. The pictures were left on the camera

I would like the pictures to stay, but unfortunately the photobooth does not start if there is a memory card in the camera slot.

best regards

chrissisdl, 2020-02-16:
Hello, I seem to have been able to solve this problem.
Now I can put my memory card in the camera and the photobooth starts without problems. The images are also saved on the camera :)

Could find something in the closed inquiries that is interesting for me.

Nevertheless many thanks

https://github.com/philipptrenz/photo-booth/issues/61

andi34, 2020-02-26:
Deletion should be avoided as mentioned in the linked Issue, it's enough to remove the rights for it 
```
sudo chmod -x /usr/lib/gvfs/gvfs-gphoto2-volume-monitor
```
Now reboot or unmount your camera with the following commands (you get a list of mounted cameras with `gio mount -l`):
```
gio mount -u gphoto2://YOUR-CAMERA
```

closed 2020-10-23 by chrissisdl